brocaar/chirpstack-application-server #100: gRPC calls return "unknown service" error

xapon, 2017-06-21:
I'm using lora-app-server v0.10.0.
When trying to call any gRPC method as described here https://docs.loraserver.io/lora-app-server/integrate/grpc/, I'm getting an error:
`rpc error: code = Unimplemented desc = unknown service api.DownlinkQueue`
(in this example, I'm calling DownlinkQueue service)
There are no logs generated on lora-app-server side when this error happens.

brocaar, 2017-06-22:
Could you share your code?

xapon, 2017-06-22:
Sure. This example completely repeats the one from the docs, except for lora-app-server url:
```
package main

import (
	"context"
	"log"

	"github.com/brocaar/lora-app-server/api"
	"google.golang.org/grpc"
)

func main() {
	// allow insecure / non-tls connections
	grpcOpts := []grpc.DialOption{
		grpc.WithInsecure(),
	}

	// connect to lora-app-server
	asConn, err := grpc.Dial("loraserver.xm:8001", grpcOpts...)
	if err != nil {
		log.Fatal(err)
	}

	// create a new node - doesn't matter what service I call, each returns the same error
	nodeAPI := api.NewNodeClient(asConn)
	_, err = nodeAPI.Create(context.Background(), &api.CreateNodeRequest{
		DevEUI:                 "0101010101010101",
		AppEUI:                 "0202020202020202",
		AppKey:                 "03030303030303030303030303030303",
		ApplicationID:          1,
		UseApplicationSettings: true,
	})
	if err != nil {
		log.Fatal(err)
	}
}
```

Execution gives this error:
`
rpc error: code = Unimplemented desc = unknown service api.Node
`

I'm sure the service is available at this url. I'm using REST API now instead of gRPC, and it's working fine.
In case this matters, I'm running lora-app-server in docker container based on this project https://github.com/ceratop/loraserver-docker but with updated components and exposed port 8001.

simonnilsson, 2017-08-28:
Hi, I'm having a similar issue. I have lora-app-server running on a Raspberry Pi. Im trying to connect using the node.js grpc package. 

```
const grpc = require('grpc');
const path = require('path');

const userProto = grpc.load({
  root: path.join(__dirname, 'proto'),
  file: 'user.proto'
}).api;

let internalApiClient = new userProto.Internal('localhost:8001', grpc.credentials.createInsecure());

internalApiClient.login({ username: 'admin', password: 'admin' }, function(err, result) {
  console.log(err);
  console.log(result);
});
```
Output:
```
{ Error: unknown service api.Internal
    at /home/pi/application/node_modules/grpc/src/node/src/client.js:569:15 code: 12, metadata: Metadata { _internal_repr: {} } }
undefined
```

brocaar, 2017-08-28:
@simonnilsson try port `8080` :-)

simonnilsson, 2017-08-28:
Okay so grpc runs on the same port as the Rest API? I guess the documentation confused me. What is running on port 8001 then?

Changing to port 8080 i get this error instead:
```
{ Error: Endpoint read failed
    at /home/pi/application/node_modules/grpc/src/node/src/client.js:569:15 code: 14, metadata: Metadata { _internal_repr: {} } }
undefined

```

brocaar, 2017-08-28:
There are two gRPC interfaces. One for the network-server to communicate with the application-server. This is running on port 8001. The other gRPC interface (running on the same port as the REST api) is running by default on port 8080. 

I think you can remove `grpc.credentials.createInsecure()` as this port uses TLS.

simonnilsson, 2017-08-28:
Okay that explains it. So with these changes it works:

```
const grpc = require('grpc');
const path = require('path');
const fs = require('fs');
const cert = fs.readFileSync('/etc/lora-app-server/certs/http.pem');

const userProto = grpc.load({
  root: path.join(__dirname, 'proto'),
  file: 'user.proto'
}).api;

let internalApiClient = new userProto.Internal('localhost:8080', grpc.credentials.createSsl(cert));

internalApiClient.login({ username: 'admin', password: 'admin' }, function(err, result) {
  console.log(err);
  console.log(result);
});
```
Thanks! :-)

xapon, 2017-08-29:
Great it's finally solved. Maybe it's worth mentioning this in documentation more clearly :)

closed 2017-08-29 by xapon

brocaar, 2017-08-29:
I've created an issue regarding the documentation improvements ^^. If you would have the time to help out by creating a pull-request, that would be great!