brocaar/chirpstack-application-server #101: Store Downlink Item in Redis instead of in PostgreSQL

mikumaycry, 2017-06-22:
While I try to send downlink msg by mosquitto, sometimes **GetDownlinkDataDelay** is not long enough to fetch the msg. And increasing **GetDownlinkDataDelay** will take up more RX Delay.

Downlink item is inserted into database asynchronously, when  **GetDataDown** operation failed to fetch a downlink item, the item will be taken by the next **GetDataDown** .

Since lora-app-server always inserts unconfirmed downlink item into database and delete it after sending  data down, is it better to store downlink item in Redis?

brocaar, 2017-06-30:
Could you tell me which problem this would solve? Personally I think it is better to keep this in PostgreSQL as it allows you to sort, filter etc... For this usage, I think PostgreSQL is a better fit :-)

mikumaycry, 2017-07-03:
PostgreSQL is good for managing downlink msg.But it's too slow to first insert a downlink msg into it and then select msg from it.
In my test, I send a confirmed msg every 20 seconds and expect a timestamp response.
My application subscribes **"application/+/node/+/rx"**. Everytime it receives a msg, it generates a timetsamp and publishes it. 
Sometimes loraserver fetchs reponse before msg is inserted into PostgreSQL.So the payload will be empty, only ACK msg is returned. Yet the msg has been insert into PostgreSQL, which makes lora-app-server always return **the old msg** in PostgreSQL util all msg in PostgreSQL are emptied.
Since downlink msg is already filtered in **txPayloadHandler**, I think lora-app-server just need to log downlink msg.

brocaar, 2017-07-03:
This has more to do with timing and the fact that receiving an uplink and sending a downlink are not directly related to each other. When you're fast enough, it will be handled in one "transaction", when you're too late, the downlink is enqueued. Potentially you'll have the same issue with Redis.

However I do see your issue where you want to make sure that a downlink is related to a received uplink. Let me think about this.

brocaar, 2017-08-21:
There might be some changes in the way this will be handled when moving towards LoRaWAN 1.1 / the way how components will interact with each other. Until then I'll park this issue.

brocaar, 2017-11-19:
In one of the next versions, I'll refactor the downlink queue. However, moving these to Redis is not going to work with Class-B due to scheduling on ping slots. Redis is not a great fit for this as I need to be able to perform queries filtered on timestamp. I'm going to close this issue.

closed 2017-11-19 by brocaar