brocaar/chirpstack-application-server #104: Send notification over MQTT when message is sent to device

xapon, 2017-06-27:
It would be useful to have a feature like SendACKNotification (in internal/handler/mqtt_handler.go), but for the case when message is sent and not yet confirmed by device.

My current use case for it is to aggregate and analyze statistical data on message queue, e.g. how long do messages stay in queue, the average/min/max queue size and so on. 
My current workaround is to poll a REST API periodically to get pending messages, but it's not precise nor convenient.

To reduce possible overhead, this feature may be optional.

If you don't mind to have this feature, I would be glad to implement it myself. It would behave similarly to ack notification, and the main code changes would be to internal/api/application_server.go around [here](https://github.com/brocaar/lora-app-server/blob/master/internal/api/application_server.go#L358) and to internal/handler/mqtt_handler.go (like the SendACKNotification method).

brocaar, 2017-12-11:
Sorry for replying so late in this issue. Please note that for correct behavior, this would require an ACK from the gateway that the frame has been enqueued for transmission (which is not yet implemented).

Also I released a new release with a queue refactor. The data will now be stored in the LoRa Server database as the queue logic should be more in the domain of the network-server (and it makes it easier to implement Class-B in the future). In the new `device_queue` table, there will be some extra meta-data stored which could potentially be used for analytics.

brocaar, 2020-01-23:
This has been implemented and merged into the master branch. Will be included in the next release.

closed 2020-01-23 by brocaar