brocaar/chirpstack-application-server #107: unable to start lora-app-server

Ankitjd18, 2017-06-30:
Hi,
I was trying to run the server but getting below error
`gwmigrate: list gateways error (will retry): rpc error: code = 14 desc = grpc: the connection is unavailable`
Please help me here.
Thanks,
Ankit

glederer007, 2017-07-05:
Hello,

Hav a look at: https://github.com/brocaar/lora-app-server/issues/80

It may help you.

Ankitjd18, 2017-07-07:
I have checked the issues earlier also, I cross checked and I am using two separate dbs for servers, but still getting the same errors. I must be missing something silly.

brocaar, 2017-07-07:
Are you sure LoRa Serve is up and running? It seems that LoRa App Server can't communicate with LoRa Server.

Ankitjd18, 2017-07-08:
Yes, LoRa server is not starting and giving below error -
`time="2017-07-08T08:37:33Z" level=info msg="migrated node-sessions to new format" errors=0 migrated=0 
time="2017-07-08T08:37:33Z" level=info msg="applying database migrations" 
time="2017-07-08T08:37:33Z" level=fatal msg="applying migrations failed: pq: type "jsonb" does not exist handling 0002_node_frame_log.sql" `

brocaar, 2017-07-10:
Which version of PostgreSQL are you using?

Ankitjd18, 2017-07-10:
psql (PostgreSQL) 9.3.17

brocaar, 2017-07-10:
You need at least 9.5+, please see https://docs.loraserver.io/lora-app-server/install/requirements/

closed 2017-07-10 by brocaar

Ankitjd18, 2017-07-10:
Getting same error with
psql (PostgreSQL) 9.6.3

brocaar, 2017-07-10:
What do you get when you connect to the database and run (SQL query):

```sql
select version();
```

That should validate that you did not only upgrade the PostgreSQL client (psql), but also the database / service.

Ankitjd18, 2017-07-10:
My bad, I upgraded it completely. But, now I am getting below error in LoRa Server as well as LoRa App Server's logs -
`time="2017-07-10T15:18:08Z" level=error msg="ping database error, will retry in 2s: dial tcp 127.0.0.1:5432: getsockopt: connection refused" `
I can't see anything connecting to port 5432 in configuration files.

Ankitjd18, 2017-07-10:
Postgresql was running on 5433 port changed it back to 5432. Now it's working.
Thanks for your help.

Sadashiv97, 2019-08-29:
WARN[0018] storage: **ping PostgreSQL database error, will retry in 2s  error="dial tcp 127.0.0.1:5432: connect: connection refused"**
 what should i do Sir?

andreaspeters, 2020-02-19:
@Sadashiv97 did u already found out what the problem was?  I have the same issue.