brocaar/chirpstack-application-server #109: make build

waytai, 2017-07-07:
Building ui

> ui@0.1.0 build /home/wayne/gosrc/src/github.com/brocaar/lora-app-server/ui
> react-scripts build

module.js:471
    throw err;
    ^

Error: Cannot find module '../scripts/build'
    at Function.Module._resolveFilename (module.js:469:15)
    at Function.resolve (internal/module.js:27:19)
    at Object.<anonymous> (/home/wayne/gosrc/src/github.com/brocaar/lora-app-server/ui/node_modules/.bin/react-scripts:24:14)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:389:7)

npm ERR! Linux 4.4.0-83-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "run" "build"
npm ERR! node v6.11.0
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! ui@0.1.0 build: `react-scripts build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the ui@0.1.0 build script 'react-scripts build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the ui package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     react-scripts build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs ui
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls ui
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/wayne/gosrc/src/github.com/brocaar/lora-app-server/ui/npm-debug.log
make: *** [ui/build] Error 1

waytai, 2017-07-07:
not use docker, and use ubuntu

brocaar, 2017-07-07:
Did you run `make requirements ui-requirements`? If yes, could you try again in the docker container? If that works, please check the differences in your own setup.

See also: https://docs.loraserver.io/lora-app-server/community/source/.

waytai, 2017-07-09:
yes ,I have run make requirements ui-requirements , my ubuntu is ubuntu4.04.

brocaar, 2017-07-10:
Just tried again using the provided docker-compose / Docker container.

```bash
docker-compose build appserver
docker-compose run --rm appserver bash

# from within the container
make requirements ui-requirements
make build
```

All works fine.

Versions:

```
root@513f7470d9c6:/go/src/github.com/brocaar/lora-app-server# node --version
v7.10.0
root@513f7470d9c6:/go/src/github.com/brocaar/lora-app-server# npm --version
4.2.0
```

Could you please test also from the Docker container and then try to find out what the config differences are with your own Ubuntu based setup? It should be possible to build everything without using Docker (and I'm doing so myself), but as every setup is potentially different and with different versions, I would like to keep the docker-compose / Docker way as a reference.

waytai, 2017-07-10:
thank you. Now is ok

closed 2017-07-11 by brocaar