brocaar/chirpstack-application-server #110: Can't add new gateway -> mac adress field

D4rkness, 2017-07-07:
I have a issue.
I cant add a new gateway to over the app-server.

javasciprt line:
`id:"mac",type:"text",placeholder:"0000000000000000",pattern:"[A-Fa-f0-9]{16}",required`

javascript shows the mac-adress requieres a string of 16 chars and with the pattern.

So i can#t add a new gateway following mac adresses due to the pattern restriction

A8:40:41:16:8E:FE
or
A8-40-41-16-8E-FE

And this also doesn't work because the string has not 16 chars:
A84041168EFE

any solution for the problem ?

Thank you

brocaar, 2017-07-07:
Please note you need to use the MAC address as configured in the packet-forwarder configuration, not the ethernet MAC address. This packet-forwarder MAC address is exactly 16 chars.

D4rkness, 2017-07-07:
oh ok thank you !

closed 2017-07-07 by D4rkness