brocaar/chirpstack-application-server #113: Could it be RelaxFCnt is not set correctly in Node Session?

jjochem, 2017-07-12:
First of all: thank you for all your work on this project @brocaar. Really appreciated! 

# Background
Recently we started testing your LoRa stack to run as the network server for a small private network in the Netherlands. In our case nodes are added to the network using ABP; OTAA is no option because our nodes are located in nearly isolated environment where sending a LoRa signal is possible but receiving isn't. fCnt is not persisted on our LoRa nodes so we thought to use the relax frame-counter option to disable checking frame counters at all as described in [https://github.com/brocaar/loraserver/issues/#133](url)

# Bug?
During testing we found setting the checkboxes of 'enable frame-counter' on both the application and node levels not doing anything. Each time we get `level=error msg="processing rx packet error: get node-session error: node-session does not exist or invalid fcnt or mic"`.

# Experiment
If I understand correctly LoraAppServer should set the RelaxFCnt flag in the value of the corresponding lora:ns:session:X in Redis. This should be the input for LoraServer. Is this correct?

## Test
To explore this problem further we did the following:
1. Change values using UI according to all 2^3 = 8 combinations of the settings. (App config > Nework Settings > Enable relax frame-counter, Node configuration > Node details > Use Application Settings and Node configuration > advanced network settings > enable relax framecounter)
2. Reactivate node by pressing submit again in node activation.
3. Watch value of RelaxFCnt in corresponding lora:ns:session key

## Result
The RelaxFCnt value does not change in Redis and consistently reads: 
> \x01\x02\x00\x01\b

Any help would be appreciated.

brocaar, 2017-07-12:
Did you re-activate the node after changing the relax-framecounter setting (e.g. re-submit the node activation form for ABP)? See also the note mentioned above the form :

> Please note that changes made below (when updating a node) only have effect after (re)activating the node (either ABP or OTAA).

jjochem, 2017-07-12:
After editing the form I went to the corresponding Node in the UI > Node Activation > and pressed SUBMIT again.

Corresponding logs of LoraAppServer:
> time="2017-07-12T08:25:26Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:25:26Z" level=info msg="node updated" application_id=1 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:25:58Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:25:58Z" level=info msg="node activated" application_id=1 dev_addr=b13cc3f2 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:26:35Z" level=info msg="application updated" id=1 name=door-sensor 
time="2017-07-12T08:26:40Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:26:40Z" level=info msg="node activated" application_id=1 dev_addr=b13cc3f2 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:26:49Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:26:49Z" level=info msg="node updated" application_id=1 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:27:03Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:27:03Z" level=info msg="node updated" application_id=1 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:27:07Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:27:07Z" level=info msg="node activated" application_id=1 dev_addr=b13cc3f2 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:39:11Z" level=info msg="application updated" id=1 name=door-sensor 
time="2017-07-12T08:39:29Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:39:29Z" level=info msg="node updated" application_id=1 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:39:36Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:39:36Z" level=info msg="node activated" application_id=1 dev_addr=b13cc3f2 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:41:01Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:41:01Z" level=info msg="node updated" application_id=1 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:41:16Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:41:16Z" level=info msg="node updated" application_id=1 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:41:20Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:41:20Z" level=info msg="node activated" application_id=1 dev_addr=b13cc3f2 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:41:48Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:41:48Z" level=info msg="node updated" application_id=1 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:41:57Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:41:57Z" level=info msg="node activated" application_id=1 dev_addr=b13cc3f2 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:42:31Z" level=info msg="application updated" id=1 name=door-sensor 
time="2017-07-12T08:42:37Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:42:37Z" level=info msg="node updated" application_id=1 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:42:41Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:42:41Z" level=info msg="node activated" application_id=1 dev_addr=b13cc3f2 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:44:06Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:44:06Z" level=info msg="node updated" application_id=1 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:44:15Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:44:15Z" level=info msg="node updated" application_id=1 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:44:21Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:44:21Z" level=info msg="node activated" application_id=1 dev_addr=b13cc3f2 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:45:07Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:45:07Z" level=info msg="node updated" application_id=1 dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:45:13Z" level=info msg="node updated" dev_eui=0004a30b001bd0fa 
time="2017-07-12T08:45:13Z" level=info msg="node activated" application_id=1 dev_addr=b13cc3f2 dev_eui=0004a30b001bd0fa

jjochem, 2017-07-14:
@brocaar any tips or activities we can perform to investigate this problem further?

brocaar, 2017-07-15:
I've been doing some testing, but am unable to reproduce your issue:

* Added ABP node to LoRa App Server
* Activated the node in LoRa App Server
* Activated the node itself
* Send a couple of uplink frames, all ok
* Decreased the uplink counter on the node and send uplink frame --> `level=error msg="processing rx packet error: get node-session error: node-session does not exist or invalid fcnt or mic" data_base64="QMeA8AcAAwABaLNKfrk="`
* Enabled the relax frame-counter mode in LoRa App Server
* Re-activated the node in LoRa App Server (simply saving the activation form)
* Decreased the uplink counter on the node and send uplink frame --> `level=warning msg="frame counters reset" dev_addr=07f080c7 dev_eui=0808080808080808`

So from my side, all looks good.

jjochem, 2017-07-16:
Thank you for your time @brocaar. I will repeat your test in the coming days and will share the result here.

jjochem, 2017-07-17:
Not able to reproduce our issue at this moment. Still running some tests. Will keep you posted.

brocaar, 2017-08-15:
I'll close this issue for now. Feel free to re-open it if you're able to reproduce the issue :-)

closed 2017-08-15 by brocaar