brocaar/chirpstack-application-server #114: Downlink with mDot? Does not pop from queue?

PhatHub, 2017-07-12:
Just wondering if anyone here has tried to do a Class-A downlink with an mDot/mBed mote...
(First off... am I posting this question to the correct project?)

I've queued downlink data, and I've gotten the **Arduino LMiC** (on a 32u4 Ada-Feather) to uplink and consequently get the queued packet to pop from the server's queue. Unfortunately, the 32u4 has an error, and (as it is prone to do occasionally) won't let me update/upload my sketch changes anymore.

I had switched to working with the **mDot** (mBed) LoRaWan mote, and:
* The Lora App Server logs the incoming uplink packet, 
* ...but queued downlink packets don't get popped for transmission. 

I'm in the process right now to try to use GQRX to see if my Gateway is transmitting at all, but for now:

1: I'm assuming that the lack of queue popping means that it hasn't transmitted the downlink... Am I correct in this? 
2: Also, is there anything with mDot or LoRaWan in general (like a bit flag when preparing the packet parameters) that would cause an uplink to not allow/trigger a downlink?

Thanks! :)

PhatHub, 2017-07-14:
Okay, for some reason the downlink queue is now de-queueing, but the mDot is still not receiving data., according to the debugging UART terminal.

I created a simple python sniffer on my gateway (Mirakonta-like Raspberry Pi with MTAC-915) and I see that when the gateway is sending "pull data" (0x02) requests, I see "pull ack" (0x04) but I don't see the "pull response" (0x03) packets from the Loraserver/bridge. (I have the bridge, server, and app server running in a VM)

What would cause the transaction to have Pull ACKs, but no Pull Response?

Thanks!

siscia, 2017-07-14:
Actually I am experiencing something similar and I can't get my node to downlink any messages.
Very last release of everything.

Do we have some automatic test in place so that we are sure that the problem is on my side?
(The semtech NS seems to work just fine with exactly the same configuration of nodes and gateways.)

brocaar, 2017-07-14:
Let me validate!

brocaar, 2017-07-14:
I've been testing with the latest release of all components and all works at my side. I've been testing with both confirmed and unconfirmed. 

Could you share the following information:

* Downlink queue items are confirmed / unconfirmed
* Logs from `loraserver` process
* Logs from `lora-app-server` process
* Frame-counters from the web-interface and frame-counters from the node (both uplink and downlink)

siscia, 2017-07-14:
Unfortunately, I don't have now the hardware handy close to me, and I won't for a while.

Vacation time in Italy.

One last thing. The problem was mainly using ABP.
Brocaar, I don't want to waste your time, I believe that there was something off from our side even if I wasn't able to figure out what.

However it was weird because I was able to receive messages either in confirmed and unconfirmed way, but the confirmed one didn't yield an ACK.

(Just dumping more info.)

Cheers,

PhatHub, 2017-07-14:
@brocaar okay I'll open the issue on the lora-gateway-bridge.

Like @siscia I'm thinking maybe I have a configuration error somewhere down the line. I'll try to also post my logs. 
* The downlink queue have "confirmed" set to `true`. 
* My frame counters have been incrementing. For a while (when I couldn't get downlink packets to dequeue) the downlink counter stayed at zero, but now it's climbed to 13. Uplink counter is at 5,156.

I can't confirm with GQRX for a transmit (sometimes I see harmonics of the original uplink so I might confuse it with the downlink) but I figured the lack of Pull_Resp was an indicator.

PhatHub, 2017-07-14:
I subscribed to `gateway/#` and I am not seeing `<gateway>/tx` messages (with `txInfo` json), although the queue is de-queueing.

brocaar, 2017-07-15:
There might be only tricky thing that I need to investigate. At the LoRa Server side I'm increasing the frame-counter when the frame has been ACKed. However when the downlink was received by the node and it replied with an uplink ACK that was not received by LoRa Server, it will re-transmit the same frame again with the same frame-counter, which gets (should) rejected by the node.

This will cause a never ending loop, as LoRa Server never sees the frame acknowledged. Incrementing could be a solution, but causes side-effects like I describe in this blog post: https://medium.com/@brocaar/notes-on-lorawan-security-7e741a8ee4fa.

brocaar, 2017-07-17:
@PhatHub could you please also share some logs? Instead of the `/var/log/...` it could be that you can find your logs using the `journalctl` command.

PhatHub, 2017-07-17:
@brocaar I will get right on it... Sorry for the delay! :)

PhatHub, 2017-07-21:
As stated on the lora-gateway-bridge, I'm closing this issue (since they're related issues). 
I found evidence via GQRX that the gateway is transmitting on RX2, but I'm determining that the error is on my mDot's configuration somewhere.

As a reference, I'm dropping a link to [this mDot question](https://developer.mbed.org/questions/61950/Receive-data/) where it seems that others have had issues with getting mDot to "get" downlink transmissions.

If anyone has an mbed5-related solution, I'd appreciate it if they'd post it here. 

Thank you everyone!

closed 2017-07-21 by PhatHub

PhatHub, 2017-08-05:
Just a little question for the default US915 RX2 parameters...
I've just switched over to the Arduino LMiC. I'm assuming that the RX2 parameters are:
- 923.3Mhz (of course)
- SF=12
- BW=500
- CR=4/8?

...if I set the Rx2 datarate to 0?

Thanks. I've also widened my window timing by 10%. 
I'm sorry for being such a bother about this... The LmiC ported by matthijskooijman defaults to TheThingsNetwork rx2 values, which are "non-standard" parameters.

brocaar, 2017-08-15:
@PhatHub please use https://forum.loraserver.io for questions that are not related to features / bugs 😉 You can request the specs + all regional parameters here: https://www.lora-alliance.org/lorawan-for-developers.