brocaar/chirpstack-application-server #115: lora-app-server seems to not fully follow MQTT server restarts

Ektanoor, 2017-07-13:
In cases when the app-server publishes application data to MQTT, things may go awry if the MQTT server has been restarted. We have seen this situation twice but quite clearly. We have a server here with some 15 to 20 clients and four main topics. We restart it and everything keeps working except the publishing of 'application/#'. We reach the app server and everything seems to keep working but no messages are being sent. Only a restart of app-server seems to solve the problem.

brocaar, 2017-07-13:
Which version of LoRa App Server are you using?

Ektanoor, 2017-07-13:
10.0

Ektanoor, 2017-07-13:
Oh. As far as I can understand this issue was solved in 10.1. Is it correct? If so, sorry I didn't look directly at the versions

brocaar, 2017-07-13:
Yes, this issue was solved in 0.10.1 😄

closed 2017-07-13 by brocaar