brocaar/chirpstack-application-server #116: flushed queue

interfaith, 2017-07-13:
a large queue was flushed, attempting to see 'new' messages published.
however now no rows appear and the error shows ?

level=info msg="data-down item requested by network-server, but queue is empty" dev_eui=0000000802030402

interfaith, 2017-07-13:
the download (to the node) was working with old queue data until the queue was flushed :(

client.publish('gateway/00000000802030401/tx', '{"reference": "12345","confirmed": false ,"fPort": 1, "data": "SmF2YVNjcmlwdA==" }');

brocaar, 2017-07-14:
Well when you flush the queue, there are no more items. Please note that

> level=info msg="data-down item requested by network-server, but queue is empty" dev_eui=0000000802030402

Is not an error but an info (maybe this should be changed to debug). It is just to inform you that there was a downlink opportunity, but there was nothing to send (as the queue has been flushed / is empty).

interfaith, 2017-07-14:
yes, however a new message is published to mqtt when data from the node is received as a duplex test.
so the queue should get a new message for each data in , as it was until the flush was done.

brocaar, 2017-07-15:
Could you please test through the API that the payload you enqueued was really added to the queue? You can do this without the need to send any uplink frames from your node. In case the payload was not added to the queue, could you please check the LoRa App Server logs for any errors?

interfaith, 2017-07-15:
Yes I will re-check. I did note that there were no errors on any of the
server programs.

interfaith, 2017-07-17:
here are the 3 logs. Need to log where the queue insert is done.

Jul 17 15:44:05 jflab-gw-lora lora-app-server[1349]:
time="2017-07-17T15:44:05-07:00" level=info msg="data-down item requested
by network-server, but queue is empty" dev_eui=0000000802030402
Jul 17 15:44:21 jflab-gw-lora lora-app-server[1349]:
time="2017-07-17T15:44:21-07:00" level=info msg="handler/mqtt: publishing
data-up payload" topic="application/2/node/0000000802030402/rx"
Jul 17 15:44:21 jflab-gw-lora lora-app-server[1349]:
time="2017-07-17T15:44:21-07:00" level=info msg="data-down item requested
by network-server, but queue is empty" dev_eui=0000000802030402

ul 17 15:47:38 jflab-gw-lora loraserver[1346]:
time="2017-07-17T15:47:38-07:00" level=info msg="rx info sent to
network-controller" dev_eui=0000000802030402
Jul 17 15:47:38 jflab-gw-lora loraserver[1346]:
time="2017-07-17T15:47:38-07:00" level=info msg="node-session saved"
dev_addr=0138766b dev_eui=0000000802030402
Jul 17 15:47:46 jflab-gw-lora loraserver[1346]:
time="2017-07-17T15:47:46-07:00" level=info msg="backend/gateway: rx packet
received"
Jul 17 15:47:46 jflab-gw-lora loraserver[1346]:
time="2017-07-17T15:47:46-07:00" level=info msg="packet(s) collected"
dev_eui=0000000802030402 gw_count=1 gw_macs=aa555a0000000101
mtype=UnconfirmedDataUp
Jul 17 15:47:46 jflab-gw-lora loraserver[1346]:
time="2017-07-17T15:47:46-07:00" level=info msg="rx info sent to
network-controller" dev_eui=0000000802030402
Jul 17 15:47:46 jflab-gw-lora loraserver[1346]:
time="2017-07-17T15:47:46-07:00" level=info msg="node-session saved"
dev_addr=0138766b dev_eui=0000000802030402

Jul 17 15:48:33 jflab-gw-lora lora-gateway-bridge[1330]:
time="2017-07-17T15:48:33-07:00" level=info msg="gateway: sending udp
packet to gateway" addr=127.0.0.1:45538 protocol_version=1 type=PushACK
Jul 17 15:48:33 jflab-gw-lora lora-gateway-bridge[1330]:
time="2017-07-17T15:48:33-07:00" level=info msg="gateway: received udp
packet from gateway" addr=127.0.0.1:44696 protocol_version=1 type=PullData
Jul 17 15:48:33 jflab-gw-lora lora-gateway-bridge[1330]:
time="2017-07-17T15:48:33-07:00" level=info msg="gateway: sending udp
packet to gateway" addr=127.0.0.1:44696 protocol_version=1 type=PullACK
Jul 17 15:48:40 jflab-gw-lora lora-gateway-bridge[1330]:
time="2017-07-17T15:48:40-07:00" level=info msg="gateway: received udp
packet from gateway" addr=127.0.0.1:45538 protocol_version=1 type=PushData
Jul 17 15:48:40 jflab-gw-lora lora-gateway-bridge[1330]:
time="2017-07-17T15:48:40-07:00" level=info msg="gateway: sending udp
packet to gateway" addr=127.0.0.1:45538 protocol_version=1 type=PushACK
Jul 17 15:48:40 jflab-gw-lora lora-gateway-bridge[1330]:
time="2017-07-17T15:48:40-07:00" level=info msg="gateway: rxpk packet
received" addr=127.0.0.1:45538 data="QGt2OAGAeQAB5l2uc4iEvn9jVR1fx/47"
mac=aa555a0000000101
Jul 17 15:48:40 jflab-gw-lora lora-gateway-bridge[1330]:
time="2017-07-17T15:48:40-07:00" level=info msg="backend: publishing
packet" topic="gateway/aa555a0000000101/rx"
Jul 17 15:48:40 jflab-gw-lora lora-gateway-bridge[1330]:
time="2017-07-17T15:48:40-07:00" level=info msg="gateway: rxpk packet
received" addr=127.0.0.1:45538 data="QGt2OAGAeQAB5l2uc4iEvn9jVR1fx/47"
mac=aa555a0000000101
Jul 17 15:48:40 jflab-gw-lora lora-gateway-bridge[1330]:
time="2017-07-17T15:48:40-07:00" level=info msg="backend: publishing
packet" topic="gateway/aa555a0000000101/rx"
Jul 17 15:48:44 jflab-gw-lora lora-gateway-bridge[1330]:
time="2017-07-17T15:48:44-07:00" level=info msg="gateway: received udp
packet from gateway" addr=127.0.0.1:44696 protocol_version=1 type=PullData
Jul 17 15:48:44 jflab-gw-lora lora-gateway-bridge[1330]:
time="2017-07-17T15:48:44-07:00" level=info msg="gateway: sending udp
packet to gateway" addr=127.0.0.1:44696 protocol_version=1 type=PullACK

interfaith, 2017-07-18:
Maybe the "api" will work instead of "mqtt" publishing ?

Perhaps there is a debug level to show more transactions of mqtt published.

brocaar, 2017-07-18:
It should print a log message, both when enqueueing through MQTT and the API. It seems the downlink queue payload was not received / processed by LoRa App Server.

interfaith, 2017-07-18:
this seems to be working now, thank-you.

client.publish('application/....) works !
client.publish('gateway/...) did not
The documentation mentions both ways.

client.publish('application/2/node/0000000802030402/tx', '{"reference":
"12345","confirmed": true,"fPort": 1, "data": ' + '"' + bigbuf + '"  }');

//client.publish('gateway/00000000802030401/tx', '{"reference":
"12345","confirmed": false ,"fPort": 1, "data": "SmF2YVNjcmlwdA==" }');

//client.publish('gateway/00000000802030401/tx', '{"reference":
"12345","confirmed": false ,"fPort": 1, "data": ' + '"' + bigbuf + '"  }');

//client.publish('gateway/00000000802030402/tx', '{"reference":
"12345","confirmed": false ,"fPort": 1, "data": ' + '"' + bigbuf + '"  }');

18 07:46:26 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:26-07:00" level=info msg="handler/mqtt: publishing
data-up payload" topic="application/2/node/0000000802030402/rx"
Jul 18 07:46:26 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:26-07:00" level=info msg="handler/mqtt: data-down
payload received" topic="application/2/node/0000000802030402/tx"
Jul 18 07:46:26 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:26-07:00" level=info msg="downlink queue item
enqueued" dev_eui=0000000802030402 id=730
Jul 18 07:46:26 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:26-07:00" level=info msg="downlink queue item
updated" id=725
Jul 18 07:46:26 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:26-07:00" level=info msg="data-down item requested
by network-server" confirmed=true dev_eui=0000000802030402 fcnt=0 id=725
Jul 18 07:46:41 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:41-07:00" level=info msg="handler/mqtt: publishing
data-up payload" topic="application/2/node/0000000802030402/rx"
Jul 18 07:46:41 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:41-07:00" level=info msg="handler/mqtt: data-down
payload received" topic="application/2/node/0000000802030402/tx"
Jul 18 07:46:41 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:41-07:00" level=info msg="downlink queue item
enqueued" dev_eui=0000000802030402 id=731
Jul 18 07:46:41 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:41-07:00" level=info msg="downlink queue item
updated" id=725
Jul 18 07:46:41 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:41-07:00" level=info msg="data-down item requested
by network-server" confirmed=true dev_eui=0000000802030402 fcnt=0 id=725
Jul 18 07:46:49 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:49-07:00" level=info msg="handler/mqtt: publishing
data-up payload" topic="application/2/node/0000000802030402/rx"
Jul 18 07:46:49 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:49-07:00" level=info msg="handler/mqtt: data-down
payload received" topic="application/2/node/0000000802030402/tx"
Jul 18 07:46:49 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:49-07:00" level=info msg="downlink queue item
enqueued" dev_eui=0000000802030402 id=732
Jul 18 07:46:49 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:49-07:00" level=info msg="downlink queue item
updated" id=725
Jul 18 07:46:49 jflab-gw-lora lora-app-server[1349]:
time="2017-07-18T07:46:49-07:00" level=info msg="data-down item requested
by network-server" confirmed=true dev_eui=0000000802030402 fcnt=0 id=725

brocaar, 2017-07-18:
> this seems to be working now, thank-you.

Great!

> client.publish('application/....) works !
> client.publish('gateway/...) did not
> The documentation mentions both ways.

Could you add links to the documentation? What the documentation should communicate is that to let your gateway send raw LoRa(WAN) frames, you would use the `gateway/[mac]/tx` topic. This is useful when you just want to communicate with your gateway and bypass the network-server (or maybe implement your own network-server). 

The `application/...` topic should be used when enqueueing data for a given node in your network.

Please let me know if this was just a misunderstanding, or if this could be improved by updating the documentation.

interfaith, 2017-07-19:
"Hello World" examples are "worth a 1000 words".  Getting data back to the
node made sense
via  application/....topic . However some syntax error likely made me try
the 2nd
method via gateway/. What puzzled me is how the queue could build up ?
Perhaps if the server or node is restarted OTAA must be redone.
Somehow a huge queue built up.?

The message queue grows as it is not confirmed ?  So the message
is sent over and over ? The node postgres table has a "pending" flag.
This pending flag is sometimes true sometimes false.

So it seems the node is not confirming the message sent.

interfaith, 2017-07-19:
Here is some output from the node, claiming unconfirmed type message.
So why is the message not removed from the queue ??

RaMac.c->onRadioTXDone] RX2 SET Timer3 at RTC:[66864459][1797] SendFrame OK

[RxWindow 1]LoRaMacParams.ChannelsDatarate:[0] rxConfig.Datarate:[0]
rxConfig.DrOffset:[0]
FRAME_TYPE_DATA_(UN)CONFIRMED_DOWN

brocaar, 2017-07-20:
That is exactly what `confirmed: false` does. As soon as the frame is transmitted, it is popped from the queue. However when there is a `confirmed: true` frame in the queue, it will stay there until an ACK is received by the node.

interfaith, 2017-07-20:
Great so now confirmed: false  
yet the message repeats

interfaith, 2017-07-20:
it works ! had to flush truncate the download queue.
yet with confirmed: true the node says it is an unconfirmed message

anyway now with unconfirmed it appears to be working not repeating , yeah !

interfaith, 2017-07-20:
confirmed: false works !
confirmed: true -> node still says "unconfirmed' -> queue grows

interfaith, 2017-07-20:
Yes, great. However the node is still seeing the message as of type
"unconfirmed".
It is great to have "confirmed: false" working though !!

brocaar, 2017-07-21:
Maybe that is an issue with the node firmware? Could you make a screenshot of the frame-log (the last downlink sent) that you find in the web-interface? Debugging this from distance is a bit difficult as I don't see the complete picture... Maybe you could also send me a database dump of your LoRa Server and LoRa App Server database to my e-mail (info [at] brocaar [dot] com). That allows me to dig a little bit deeper into the issue.

interfaith, 2017-07-21:
Using confirmed:true

more to come

interfaith, 2017-07-21:
more screen capture for confirmed: true repeat loop

brocaar, 2017-07-21:
@interfaith when you're replying by e-mail attachements are not added to your post.

brocaar, 2017-07-21:
Issue has been discussed outside of GitHub, seems like a node firmware issue (not ACKing confirmed downlink frames).

closed 2017-07-21 by brocaar