brocaar/chirpstack-application-server #117: Lora-app-server payload

grendy, 2017-07-17:
Hi @brocaar
Can you explain how to read data from lora-app-server gui?
I subscribe from application/+/node/+/rx mqtt broker, and i got json
{
"applicationID" : "1",
"applicationName" : "Lora",
"nodeName" : "Lora",
"devEUI" : "33393437673xxxx",
"rxInfo" : [ {
"mac" : "7276ff000b0xxxxx",
"rssi" : -62,
"loRaSNR" : 11,
"name" : "Gateway",
"latitude" : -x.xxxxx,
"longitude" : xxx.xxxxx,
"altitude" : 40
} ],
"txInfo" : {
"frequency" : 923200000,
"dataRate" : {
"modulation" : "LORA",
"bandwidth" : 125,
"spreadFactor" : 10
},
"adr" : true,
"codeRate" : "4/5"
},
"fCnt" : 726,
"fPort" : 2,
"data" : "CPw="
}

GUI lora-app-server:
[image]

Thanks before. #

brocaar, 2017-07-17:
The data in your screenshot is the RAW and encrypted data (e.g. how it is transmitted over the air). What you see in the JSON is the unencrypted data.

closed 2017-07-17 by brocaar

Deepdone, 2017-07-18:
@brocaar How can i get the real unencrypted data through The data in @grendy's screenshot(bytes:"2l8=")

brocaar, 2017-07-18:
The data published to the `application/+/node/+/rx` MQTT topic is decrypted.

Deepdone, 2017-07-18:
@brocaar  i want to use the Rest API ,and decrypt data instead of MQTT.the way is more convenient,but It doesn't seem to be using BASE64 encryption.

brocaar, 2017-07-18:
The (raw) frame log is not meant for integration with your application, it is more for debugging what is going over the wire (like you would use Wireshark for example).

Please wait until https://github.com/brocaar/lora-app-server/pull/102 has been merged in (I expect this will merged in next week or the week after). It will allow you to setup HTTP endpoints to receive data as an alternative to MQTT. The JSON structure will be exactly the same as the JSON over MQTT.

To decrypt `2l8=` from the above screenshot, you could use this function (the encrypt function is used for both encryption / decryption): https://godoc.org/github.com/brocaar/lorawan#EncryptFRMPayload.