brocaar/chirpstack-application-server #118: No Frame Logs

DAprojectlora, 2017-07-19:
Hello, i'm testing the Loraserver, but i got some issues:
I am working with Microchip node and gateway, and loraserver.io files on windows.

Main issue is on the web browser, the Frame logs is empty, even if uplink frame-counter is working (up when I send msg)

It can be link:

 I can't use API because I got "authentication failed: get token from context error: no authorization-data in metadata"
I can get a token from /api/internal/login , but past in top-right corner, i got "fetching resource list: undefined; Please wait.

During confirmed msg, i got Data Transmitted on the mote, up of Uplink frame counter and Downlink frame counter but got an 
"processing rx packet error: unknown MType: UnconfirmedDataDown" on the Loraserver console

Maybe you can help me with that

DAprojectlora, 2017-07-19:
I solved authentication issue, #41 
However i also get:
"processing rx packet error: unknown MType: JoinAccept" (and device denied otaa)

brocaar, 2017-07-20:
Looks like the gateway is receiving its own data (or data from a gateway really close by). Or the node sends the wrong message type.

DAprojectlora, 2017-07-20:
Seems the whole thing is working even i got this message sometimes,
so is it normal to have frame logs empty?
It can receive the data by mosquitto_sub cmd.
Now i ll try to extract data, i saw http handler required.

brocaar, 2017-07-25:
Frame logging is a feature that you can enable in LoRa Server, see also: https://docs.loraserver.io/loraserver/install/config/.

brocaar, 2017-07-26:
I'm closing this issue. In case you would like to discuss it any further, please post your question at https://forum.loraserver.io/ (you should be able to login with your GitHub account).

closed 2017-07-26 by brocaar