brocaar/chirpstack-application-server #119: grpc issue prevails in lora server

monkmahadeva, 2017-07-21:
Hi brocaar I have been trying to Configure the LORA environment as mentioned in https://docs.loraserver.io/install/quick-install/ ,Its been really handy as of now.I did encounter a few errors while i was setting it up which however i resolved in due time. MQTT has been configured for all the three components(Lora-gateway-bridge, Loraserver, Lora-app-server). The Lora-app-server is up and running in the HTTP_BIND with this pretty web interface as specified in the docs but with a security exception which prompted me to get my own Certificate through the browser itself.So firstly one doubt i have is According to the docs once the  Lora App Server is running, this error should stop:
 INFO[0001] grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 0.0.0.0:8000: getsockopt: connection refused"; Reconnecting to {0.0.0.0:8000 <nil>}.
 I neglected this initially since my Lora-app-server was working fine, But now i,m trying to retrieve the UDP payload from Lora-gateway configured with a packet forwarder to the Lora-gateway-bridge. But Even though i have burned the midnight oil for this i am still stuck. Lol..Here is the journal of all the three components.
Lora-gateway-bridge:
       -- Logs begin at Thu 2017-07-20 03:55:11 PDT. --
Jul 20 03:55:40 ubuntu systemd[1]: Started LoRa Gateway Bridge.
Jul 20 03:55:42 ubuntu lora-gateway-bridge[1186]: time="2017-07-20T03:55:42-07:00" level=info msg="starting LoRa Gateway Bridge" docs="https://docs.loraserver.io/lora-gateway-bridge/" version=2.1.5
Jul 20 03:55:42 ubuntu lora-gateway-bridge[1186]: time="2017-07-20T03:55:42-07:00" level=info msg="backend: connecting to mqtt broker" server="tcp://127.0.0.1:1883"
Jul 20 03:55:42 ubuntu lora-gateway-bridge[1186]: time="2017-07-20T03:55:42-07:00" level=info msg="backend: connected to mqtt broker"
Jul 20 03:55:42 ubuntu lora-gateway-bridge[1186]: time="2017-07-20T03:55:42-07:00" level=info msg="gateway: starting gateway udp listener" addr=127.0.0.1:1700
Jul 20 21:59:54 ubuntu systemd[1]: Stopping LoRa Gateway Bridge...
Jul 20 21:59:54 ubuntu lora-gateway-bridge[1186]: time="2017-07-20T21:59:54-07:00" level=info msg="signal received" signal=terminated
Jul 20 21:59:54 ubuntu lora-gateway-bridge[1186]: time="2017-07-20T21:59:54-07:00" level=warning msg="shutting down server"
Jul 20 21:59:54 ubuntu lora-gateway-bridge[1186]: time="2017-07-20T21:59:54-07:00" level=info msg="gateway: closing gateway backend"
Jul 20 21:59:54 ubuntu lora-gateway-bridge[1186]: time="2017-07-20T21:59:54-07:00" level=info msg="gateway: handling last packets"
Jul 20 21:59:55 ubuntu systemd[1]: Stopped LoRa Gateway Bridge.
Jul 20 21:59:55 ubuntu systemd[1]: Started LoRa Gateway Bridge.
Jul 20 21:59:55 ubuntu lora-gateway-bridge[4382]: time="2017-07-20T21:59:55-07:00" level=info msg="starting LoRa Gateway Bridge" docs="https://docs.loraserver.io/lora-gateway-bridge/" version=2.1.5
Jul 20 21:59:55 ubuntu lora-gateway-bridge[4382]: time="2017-07-20T21:59:55-07:00" level=info msg="backend: connecting to mqtt broker" server="tcp://127.0.0.1:1883"
Jul 20 21:59:56 ubuntu lora-gateway-bridge[4382]: time="2017-07-20T21:59:56-07:00" level=info msg="gateway: starting gateway udp listener" addr=127.0.0.1:1700
Jul 20 21:59:56 ubuntu lora-gateway-bridge[4382]: time="2017-07-20T21:59:56-07:00" level=info msg="backend: connected to mqtt broker"

Loraserver:

-- Logs begin at Thu 2017-07-20 03:55:11 PDT. --
Jul 20 03:55:40 ubuntu systemd[1]: Started LoRa Server.
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:42-07:00" level=info msg="starting LoRa Server" band="EU_863_870" docs="https://docs.loraserver.io/" net_id=010203 version=0.19.2
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:43-07:00" level=info msg="setup redis connection pool" url="redis://localhost:6379"
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:43-07:00" level=info msg="connecting to postgresql"
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:43-07:00" level=info msg="backend/gateway: connecting to mqtt broker" server="tcp://localhost:1883"
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:43-07:00" level=info msg="connecting to application-server" ca-cert= server="127.0.0.1:8001" tls-cert= tls-key=
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:43-07:00" level=info msg="backend/gateway: connected to mqtt server"
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:43-07:00" level=info msg="backend/gateway: subscribing to rx topic" topic="gateway/+/rx"
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:43-07:00" level=info msg="no network-controller configured"
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:43-07:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:8001: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8001 <nil>}"
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:43-07:00" level=info msg="migrated node-sessions to new format" errors=0 migrated=0
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:43-07:00" level=info msg="applying database migrations"
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:43-07:00" level=info msg="backend/gateway: subscribing to stats topic" topic="gateway/+/stats"
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:43-07:00" level=info msg="migrations applied" count=0
Jul 20 03:55:43 ubuntu loraserver[1188]: time="2017-07-20T03:55:43-07:00" level=info msg="starting api server" bind="localhost:8000" ca-cert= tls-cert= tls-key=
Jul 20 03:55:44 ubuntu loraserver[1188]: time="2017-07-20T03:55:44-07:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:8001: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8001 <nil>}"
Jul 20 03:55:46 ubuntu loraserver[1188]: time="2017-07-20T03:55:46-07:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:8001: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8001 <nil>}"
Jul 20 04:01:12 ubuntu loraserver[1188]: time="2017-07-20T04:01:12-07:00" level=info msg="gateway created" mac=b827ebffff9ccc85
Jul 20 22:01:41 ubuntu systemd[1]: Stopping LoRa Server...
Jul 20 22:01:41 ubuntu loraserver[1188]: time="2017-07-20T22:01:41-07:00" level=info msg="signal received" signal=terminated
Jul 20 22:01:41 ubuntu loraserver[1188]: time="2017-07-20T22:01:41-07:00" level=warning msg="stopping loraserver"
Jul 20 22:01:42 ubuntu loraserver[1188]: time="2017-07-20T22:01:41-07:00" level=info msg="backend/gateway: closing backend"
Jul 20 22:01:42 ubuntu loraserver[1188]: time="2017-07-20T22:01:42-07:00" level=info msg="backend/gateway: unsubscribing from rx topic" topic="gateway/+/rx"
Jul 20 22:01:43 ubuntu loraserver[1188]: time="2017-07-20T22:01:43-07:00" level=info msg="backend/gateway: unsubscribing from stats topic" topic="gateway/+/stats"
Jul 20 22:01:43 ubuntu loraserver[1188]: time="2017-07-20T22:01:43-07:00" level=info msg="backend/gateway: handling last messages"
Jul 20 22:01:43 ubuntu loraserver[1188]: time="2017-07-20T22:01:43-07:00" level=info msg="waiting for pending actions to complete"
Jul 20 22:01:43 ubuntu systemd[1]: Stopped LoRa Server.
Jul 20 22:01:43 ubuntu systemd[1]: Started LoRa Server.
Jul 20 22:01:44 ubuntu loraserver[4404]: time="2017-07-20T22:01:44-07:00" level=info msg="starting LoRa Server" band="EU_863_870" docs="https://docs.loraserver.io/" net_id=010203 version=0.19.2
Jul 20 22:01:44 ubuntu loraserver[4404]: time="2017-07-20T22:01:44-07:00" level=info msg="setup redis connection pool" url="redis://localhost:6379"
Jul 20 22:01:44 ubuntu loraserver[4404]: time="2017-07-20T22:01:44-07:00" level=info msg="connecting to postgresql"
Jul 20 22:01:45 ubuntu loraserver[4404]: time="2017-07-20T22:01:45-07:00" level=info msg="backend/gateway: connecting to mqtt broker" server="tcp://localhost:1883"
Jul 20 22:01:45 ubuntu loraserver[4404]: time="2017-07-20T22:01:45-07:00" level=info msg="connecting to application-server" ca-cert= server="127.0.0.1:8001" tls-cert= tls-key=
[... cut ...]
Jul 20 22:01:54 ubuntu lora-app-server[4417]: time="2017-07-20T22:01:54-07:00" level=info msg="starting client api server" bind="localhost:8080" tls-cert="/etc/lora-app-server/certs/http.pem" tls-key="/etc/lora-app-server/certs/http-key.pem"
Jul 20 22:01:54 ubuntu lora-app-server[4417]: time="2017-07-20T22:01:54-07:00" level=info msg="registering rest api handler and documentation endpoint" path="/api"

Could you please tell me if there is something wrong in the configuration part.?

P.S: I am New to this so i'm having a hard time to test or debug the configuration.Cheers.

brocaar, 2017-07-21:
The above looks fine to me. After starting the LoRa App Server service, this was the last LoRa Server log:

```
Jul 20 22:01:53 ubuntu loraserver[4404]: time="2017-07-20T22:01:53-07:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:8001: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8001 <nil>}"
```

So after that it was connected.

Was that your question?

monkmahadeva, 2017-07-21:
Indeed that was one of my doubts..Thanks..Although my real concern is how to get the packet from my gateway, The gateway has been purchased from RisingHF named as RHF2S001. Since we are using channel frequency EU868, same frequency has been set in the Band Parameter of Lora Server config file.  I am able to get packet from lora_mote to gateway also the packet is being transmitted to specified server address, which is the host name of lora gateway bridge(0.0.0.0). In global_config_EU868 file of gateway i replaced the server address to 0.0.0.0 and port up and down is 1700 and in the gateway log we are able to see the packet but in Lora gateway bridge log we are not able to receive any packet. I am attaching my gateway and lora gateway bridge log report.

rxhf@rhf2s001:~ $ sudo systemctl enable pktfwd
rxhf@rhf2s001:~ $ sudo systemctl restart pktfwd
rxhf@rhf2s001:~ $ sudo systemctl status pktfwd
? pktfwd.service - packet forwarder
   Loaded: loaded (/lib/systemd/system/pktfwd.service; enabled)
   Active: active (running) since Thu 2017-07-20 13:54:18 UTC; 4s ago
  Process: 2935 ExecStartPre=/home/rxhf/risinghf/pktfwd/update_gwid.sh /home/rxhf/risinghf/pktfwd/local_conf.json (code=exited, status=0/SUCCESS)
  Process: 2927 ExecStartPre=/usr/local/sbin/gwrst (code=exited, status=0/SUCCESS)
 Main PID: 2948 (pktfwd)
   CGroup: /system.slice/pktfwd.service
           +-2948 /home/rxhf/risinghf/pktfwd/pktfwd

Jul 20 13:54:18 rhf2s001 pktfwd[2948]: lgw_rxif_setconf:572: Note: LoRa 'multi' if_chain 4 configuration; en:1 freq:-200000 SF_mask:0x7e
Jul 20 13:54:18 rhf2s001 pktfwd[2948]: lgw_rxif_setconf:572: Note: LoRa 'multi' if_chain 5 configuration; en:1 freq:0 SF_mask:0x7e
Jul 20 13:54:18 rhf2s001 pktfwd[2948]: lgw_rxif_setconf:572: Note: LoRa 'multi' if_chain 6 configuration; en:1 freq:200000 SF_mask:0x7e
Jul 20 13:54:18 rhf2s001 pktfwd[2948]: lgw_rxif_setconf:572: Note: LoRa 'multi' if_chain 7 configuration; en:1 freq:400000 SF_mask:0x7e
Jul 20 13:54:18 rhf2s001 pktfwd[2948]: lgw_rxif_setconf:546: Note: LoRa 'std' if_chain 8 configuration; en:1 freq:-200000 bw:2 dr:2
Jul 20 13:54:18 rhf2s001 pktfwd[2948]: lgw_rxif_setconf:602: Note: FSK if_chain 9 configuration; en:1 freq:300000 bw:3 dr:50000 (50000 real dr) sync:0xC194C1
Jul 20 13:54:19 rhf2s001 pktfwd[2948]: lgw_start:788: Note: calibration started (time: 2300 ms)
Jul 20 13:54:21 rhf2s001 pktfwd[2948]: lgw_start:809: Note: calibration finished (status = 191)
Jul 20 13:54:21 rhf2s001 pktfwd[2948]: Info: Initialising AGC firmware...
Jul 20 13:54:21 rhf2s001 pktfwd[2948]: Info: putting back original RADIO_SELECT value

rxhf@rhf2s001:~/risinghf/pktfwd $ ln -sf global_conf_eu868.json global_conf.json
rxhf@rhf2s001:~/risinghf/pktfwd $ sudo systemctl restart pktfwd

rxhf@rhf2s001:~ $ cd /home/rxhf/loriot/1.0.2
rxhf@rhf2s001:~/loriot/1.0.2 $ sudo systemctl stop pktfwd
rxhf@rhf2s001:~/loriot/1.0.2 $ sudo gwrst
Gateway reset successfully
rxhf@rhf2s001:~/loriot/1.0.2 $ ./lrt -f -i eth0 -s ap3.loriot.io
[INFO ]              main.c:178 LORIOT.io Gateway Version DEV.431

[LOG  ]        loraConfig.c:84  Interface eth0 found with address 192.168.0.94
[LOG  ]        loraConfig.c:99  Interface eth0 found with IP6 address fe80::3b17:4f75:a7b:dbbb
[INFO ]        loraConfig.c:143 Acquired EUI B8-27-EB-FF-FF-9C-CC-85 from interface eth0
[DEBUG]        loraConfig.c:227 Connecting config server ap3.loriot.io
[DEBUG]               tls.c:592 TLS connecting to ap3.loriot.io
[LOG  ]               tls.c:443 SSL: Will use internal certificate validation
[DEBUG]               tls.c:518 Hostname ap3.loriot.io addr 52.66.141.27
[DEBUG]               tls.c:561 Connection to ap3.loriot.io with interface eth0 192.168.0.94
[Cfg  ]        loraConfig.c:168 HTTP response  HTTP/1.1 400 Bad Request
[INFO ]        loraConfig.c:287 Missing or invalid response from gateway server -1
[ERROR]        loraConfig.c:356 Couldn't retrieve gateway configuration. Retry in 2 seconds.

[DEBUG]        loraConfig.c:227 Connecting config server ap3.loriot.io
[DEBUG]               tls.c:592 TLS connecting to ap3.loriot.io
[DEBUG]               tls.c:518 Hostname ap3.loriot.io addr 52.66.141.27
[DEBUG]               tls.c:561 Connection to ap3.loriot.io with interface eth0 192.168.0.94
[Cfg  ]        loraConfig.c:168 HTTP response  HTTP/1.1 400 Bad Request
[INFO ]        loraConfig.c:287 Missing or invalid response from gateway server -1
[ERROR]        loraConfig.c:356 Couldn't retrieve gateway configuration. Retry in 4 seconds.

[DEBUG]        loraConfig.c:227 Connecting config server ap3.loriot.io
[DEBUG]               tls.c:592 TLS connecting to ap3.loriot.io
[DEBUG]               tls.c:518 Hostname ap3.loriot.io addr 52.66.141.27
[DEBUG]               tls.c:561 Connection to ap3.loriot.io with interface eth0 192.168.0.94
[Cfg  ]        loraConfig.c:168 HTTP response  HTTP/1.1 400 Bad Request
[INFO ]        loraConfig.c:287 Missing or invalid response from gateway server -1
[ERROR]        loraConfig.c:356 Couldn't retrieve gateway configuration. Retry in 8 seconds.

[DEBUG]        loraConfig.c:227 Connecting config server ap3.loriot.io
[DEBUG]               tls.c:592 TLS connecting to ap3.loriot.io
[DEBUG]               tls.c:518 Hostname ap3.loriot.io addr 52.66.141.27
[DEBUG]               tls.c:561 Connection to ap3.loriot.io with interface eth0 192.168.0.94
[Cfg  ]        loraConfig.c:168 HTTP response  HTTP/1.1 400 Bad Request
[... cut ...]
Jul 20 21:59:56 ubuntu lora-gateway-bridge[4382]: time="2017-07-20T21:59:56-07:00" level=info msg="gateway: starting gateway udp listener" addr=127.0.0.1:1700
Jul 20 21:59:56 ubuntu lora-gateway-bridge[4382]: time="2017-07-20T21:59:56-07:00" level=info msg="backend: connected to mqtt broker"

Is the message at Jul 20 21:59:54 as shown below  confirming me the retrieval of packets??if not Please help me with this step I want it to show the packet recieved from my gateway

Jul 20 21:59:54 ubuntu lora-gateway-bridge[1186]: time="2017-07-20T21:59:54-07:00" level=info msg="gateway: handling last packets"

brocaar, 2017-07-21:
From your logs it looks like you're trying to use the Loriot packet-forwarder. This is not compatible with the LoRa Gateway Bridge. You must use the packet-forwarder provided by Semtech (https://github.com/lora-net/packet_forwarder, I think this is already installed in `/home/rxhf/risinghf/pktfwd/`). Update the `local_conf.json` file for that packet forwarder so that it points to the IP address of your LoRa Gateway Bridge instance and to the correct port (the default for LoRa Gateway Bridge is `1700` for up and down).

monkmahadeva, 2017-07-21:
Shykes!!  I'll look into it and get back. Thanks for your Help!

monkmahadeva, 2017-07-21:
Hi brocaar , We did the gateway configuration in the semtech packet-forwarder. It however resolved the  gateway configuration error. But After i restart the packet-forwarder it gives me an error "failed to start  concentrator". 

Prior to restarting packet forwarder The error message was not occurring. I have filled in my i.p address of Lora-gateway-bridge port up and down in local_conf.json as you mentioned earlier.The gateway_ID of local_conf.json is set to the unique MAC address of my gateway.

After running "./update_gwid local_conf.json" the message is shown that the gateway is set successfully to the given gateway ID. After typing in the command "./lora-pkt-fwd" it shows the error INVALID JSON FORMAT.

Alternatively, I tried to alter the gateway_conf in global_conf.json by specifying the same parameters as specified in local_conf.json and again typed in "./update_gwid gloabl_conf.json" which again set the gateway ID to the unique gateway to The MAC address. 

For IoT Starter Kit I've used: ./reset_lgw.sh stop ./reset_lgw.sh start

FInally after running the ./lora-pkt-fwd after updation i get the following log Errors.Hope you understand 
what i'm trying to express :P
Gateway:

rxhf@rhf2s001:~/semtech/gateway/packet_forwarder/lora_pkt_fwd $ ./lora_pkt_fwd
*** Beacon Packet Forwarder for Lora Gateway ***
Version: 3.0.0
*** Lora concentrator HAL library version info ***
Version: 4.0.1;
***
INFO: Little endian host
INFO: found global configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
INFO: lorawan_public 1, clksrc 1
INFO: antenna_gain 0 dBi
INFO: Configuring TX LUT with 16 indexes
INFO: radio 0 enabled (type SX1257), center frequency 867500000, RSSI offset -166.000000, tx enabled 1
INFO: radio 1 enabled (type SX1257), center frequency 868500000, RSSI offset -166.000000, tx enabled 0
INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7
INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to B827EBFFFE9CCC85
INFO: server hostname or IP address is configured to "localhost"
INFO: upstream port is configured to "1700"
INFO: downstream port is configured to "1700"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: found local configuration file local_conf.json, parsing it
INFO: redefined parameters will overwrite global parameters
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to FFFFFFFFFFFFFFFF
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: [main] concentrator started, packet can now be received

INFO: Disabling GPS mode for concentrator's counter...
INFO: host/sx1301 time offset=(1500642648s:202797µs) - drift=-1745150419µs
INFO: Enabling GPS mode for concentrator's counter.

##### 2017-07-21 13:11:20 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 0 (0 bytes)
# PUSH_DATA acknowledged: 0.00%
[... 107 lines cut ...]
INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7
INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to B827EBFFFE9CCC85
INFO: server hostname or IP address is configured to "localhost"
INFO: upstream port is configured to "1700"
INFO: downstream port is configured to "1700"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: found local configuration file local_conf.json, parsing it
INFO: redefined parameters will overwrite global parameters
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to FFFFFFFFFFFFFFFF
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
ERROR: [main] failed to start the concentrator

AND Lora-gateway-bridge still shows the same log as attached in my first message. :).

brocaar, 2017-07-21:
Sorry, but I can't help you with the `failed to start the concentrator`, this could be anything (and could be a hardware issue too). Please refer to the documentation of your gateway how it needs to be configured. As long as your gateway & packet-forwarder is not up and running, you won't see any interesting data in LoRa Gateway Bridge.

monkmahadeva, 2017-07-24:
Hi brocaar. Hope your doing good :). I have one basic doubt yet again. 
     => For IoT Starter Kit only, run: ./reset_lgw.sh stop ./reset_lgw.sh start
As mentioned i'm guessing Its restarting the lora gateway. But What does IoT starter kit means??..Do i have to execute these commands after the gateway configuration??

monkmahadeva, 2017-07-25:
Hi, I've resolved the concentrator problem. Although i'm not recieve any packets. I am using SSCOM to send a packet. Nothing appearing in the lora-gateway-bridge logs. Also if i wanna directly forward the packet to lora server.. what is the server adress , port up and down. ??.

brocaar, 2017-07-25:
Best way to debug this is to see if the packet-forwarder on your gateway receives any data. If not, it means your gateway doesn't see / receive any data so you have an other issue.

Do you see every x seconds this kind of log (`PullData` / `PullACK`)?

```
level=info msg="gateway: received udp packet from gateway" addr=127.0.0.1:53906 protocol_version=2 type=PullData
level=info msg="gateway: sending udp packet to gateway" addr=127.0.0.1:53906 protocol_version=2 type=PullACK
```

If not, it means that your packet-forwarder is not "connected" to the LoRa Gateway Bridge.

monkmahadeva, 2017-07-26:
Okay. So i guess the problem is that i have all my components installed in a vmware. Whereas my gateway components, like Extraputty and SSCOM are has been installed in the physical machine. So i guess this is the problem so i plan to install all my components within the VMware..  what do you think??

monkmahadeva, 2017-07-26:
And no i am not seeing that LOG as of now.

brocaar, 2017-07-26:
Yes, using VMWare could mean you need to setup port forwarding etc... as else your gateway is not able to communicate with the services running inside your VM. Please open a topic at https://forum.loraserver.io/ in case you would discuss this any further. I'm trying to cleanup the GitHub issue from support questions. I think GitHub issues are better to discuss bugs / features, e.g. things related to code changes :-)

closed 2017-07-26 by brocaar