brocaar/chirpstack-application-server #122: MQTT communication with ms azure

gradzik, 2017-07-26:
Hi,

I have a problem with configuration of lora-app-server. I use pre-compiled binaries of gateway bridge, lora server and lora-app-server. I can't connect to azure iot-hub.

lora-app-server[4128]: time="2017-07-26T08:36:52+02:00" level=info msg="handler/mqtt: connecting to mqtt broker" server="tcp://servername-iot.azure-devices.net:8883"
lora-app-server[4128]: time="2017-07-26T08:36:52+02:00" level=error msg="handler/mqtt: connecting to broker error, will retry in 2s: Network Error : %!s(<nil>)"

To connect to azure i need 4 parameters (tested with mqtt-spy and works fine):

Server URL:
servername-iot.azure-devices.net:8883

Client ID (in my this deveui):
BE7A0000000056D8

User name and Password according to ms docs:
https://docs.microsoft.com/en-us/azure/iot-hub/iot-hub-mqtt-support#using-the-mqtt-protocol-directly

Connection must be in TSL/SSL mode.

In lora-app-server configuration file I changed this parameters: MQTT_SERVER, MQTT_USERNAME, MQTT_PASSWORD but there is no Client ID to configure. Where can I find it? I think this might be a problem.

brocaar, 2017-07-26:
Thanks! Currently the ClientID can not be configured. I will add it (or when you would like to help out, please create a pull-request with these changes).

gradzik, 2017-07-26:
Thanks for a very quick response, I will think about this modification

siscia, 2017-07-26:
A while ago was added the possibility to use a certificate for authentication.

Your problem is not very different.

Take a look at the implementation that got merge, you may find it useful :)

https://github.com/brocaar/lora-app-server/pull/103/commits/479436501772aa99db59ca0bc539811c0da60b7b

brocaar, 2019-02-26:
I have added Azure Service Bus integration. This is a global integration that can be configured in the `lora-app-server.toml` configuration file (once released) and will let LoRa App Server forward events to an Azure Service Bus Queue or Topic so that this data then can be used by other applications or Azure services. This feature will be included in the next release :-) 

See https://github.com/brocaar/lora-app-server/commit/e7742860000333db6b175da3a040ed351942dd2a for more details.

Looking forward to your feedback!

--

I'm closing this issue as I think this is a better option than connecting LoRa App Server over MQTT to Azure.

closed 2019-02-26 by brocaar