brocaar/chirpstack-application-server #123: Changing RX window doesn't work

iBrick, 2017-07-26:
### Is this a bug or a feature request?
Bug
### What did you expect?
When I change 'Receive window' option I expect that receive window will be changed and so will frequency (RX1 - is the same downlink frequency as uplink frequency, RX2 - 869525)
### What happened?
In reality frequency doesn't change and it can be seen in frame log.
### What version are your using?
0.10.1
### How can your issue be reproduced?
1. Create device (I created ABP device). 
2. Send uplink, send downlink.
3. Change RX window. 
4. Send uplink, send downlink.
5. Check Frame log and see that frequency didn't change.
6. Remove device. Create it again with same DevAddr, etc (ABP) but with new RX window. 
7. Send uplink, downlink. 
You'll see that this time it is changed.
### Could you share your log output?
[image]

brocaar, 2017-07-26:
Please not that you need to re-activate your node in order for the changes to kick in. See also the note above the form:

> Please note that changes made below (when updating a node) only have effect after (re)activating the node (either ABP or OTAA).

brocaar, 2017-08-10:
Could you let me know if your issue has been solved?

iBrick, 2017-08-10:
Hi, 
Yes this worked. It is a little bit confusing about 'reactivating' in ABP mode. Because the only way to do this is deleting node and creating it again. Some devices that I've bought which work in ABP doesn't allow to change keys thus can't be 'reactivated'.

closed 2017-08-10 by iBrick