brocaar/chirpstack-application-server #127: Error grpc: the connection is unavailable (code: 14)

RogerioCassares, 2017-08-10:
### Is this a bug or a feature request?
BUG!, Please help me 

### What did you expect?
I expected to enter in gateways tab 

### What happened?
When I entered at "gateways" in lora server UI, a red line appears at thtop of the page. 
"Error grpc: the connection is unavailable (code: 14)"

### What version are your using?
I make provision yesterday on my older version
### How can your issue be reproduced?

### Could you share your log output?
 journalctl -u lora-gateway-bridge -f -n 50
-- Logs begin at Thu 2017-08-10 14:51:24 UTC. --
Aug 10 15:21:52 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:21:52Z" level=info msg="gateway: received udp packet from gateway" addr=10.0.2.2:42565 protocol_version=2 type=PushData
Aug 10 15:21:52 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:21:52Z" level=info msg="gateway: sending udp packet to gateway" addr=10.0.2.2:42565 protocol_version=2 type=PushACK
Aug 10 15:21:52 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:21:52Z" level=info msg="gateway: stat packet received" addr=10.0.2.2:42565 mac=7276ff00080801db
Aug 10 15:21:52 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:21:52Z" level=info msg="backend: publishing packet" topic="gateway/7276ff00080801db/stats"
Aug 10 15:21:58 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:21:58Z" level=info msg="gateway: received udp packet from gateway" addr=10.0.2.2:42565 protocol_version=2 type=PushData
Aug 10 15:21:58 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:21:58Z" level=info msg="gateway: rxpk packet received" addr=10.0.2.2:42565 data="gLJVAQIAET1kvdpIUIK8cThGvzPIgcM=" mac=7276ff00080801db
Aug 10 15:21:58 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:21:58Z" level=info msg="backend: publishing packet" topic="gateway/7276ff00080801db/rx"
Aug 10 15:21:58 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:21:58Z" level=info msg="gateway: sending udp packet to gateway" addr=10.0.2.2:42565 protocol_version=2 type=PushACK
Aug 10 15:21:58 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:21:58Z" level=info msg="gateway: received udp packet from gateway" addr=10.0.2.2:51990 protocol_version=2 type=PullData
Aug 10 15:21:58 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:21:58Z" level=info msg="gateway: sending udp packet to gateway" addr=10.0.2.2:51990 protocol_version=2 type=PullACK
Aug 10 15:22:08 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:22:08Z" level=info msg="gateway: received udp packet from gateway" addr=10.0.2.2:51990 protocol_version=2 type=PullData
Aug 10 15:22:08 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:22:08Z" level=info msg="gateway: sending udp packet to gateway" addr=10.0.2.2:51990 protocol_version=2 type=PullACK
Aug 10 15:22:18 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:22:18Z" level=info msg="gateway: received udp packet from gateway" addr=10.0.2.2:51990 protocol_version=2 type=PullData
Aug 10 15:22:18 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:22:18Z" level=info msg="gateway: sending udp packet to gateway" addr=10.0.2.2:51990 protocol_version=2 type=PullACK
Aug 10 15:22:21 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:22:21Z" level=info msg="gateway: received udp packet from gateway" addr=10.0.2.2:42565 protocol_version=2 type=PushData
Aug 10 15:22:21 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:22:21Z" level=info msg="gateway: rxpk packet received" addr=10.0.2.2:42565 data="gLJVAQIAET1kvdpLUIK8cThGv/uYzOI=" mac=7276ff00080801db
Aug 10 15:22:21 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:22:21Z" level=info msg="backend: publishing packet" topic="gateway/7276ff00080801db/rx"
Aug 10 15:22:21 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:22:21Z" level=info msg="gateway: sending udp packet to gateway" addr=10.0.2.2:42565 protocol_version=2 type=PushACK
Aug 10 15:22:22 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:22:22Z" level=info msg="gateway: received udp packet from gateway" addr=10.0.2.2:42565 protocol_version=2 type=PushData
Aug 10 15:22:22 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:22:22Z" level=info msg="gateway: stat packet received" addr=10.0.2.2:42565 mac=7276ff00080801db
Aug 10 15:22:22 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:22:22Z" level=info msg="backend: publishing packet" topic="gateway/7276ff00080801db/stats"
Aug 10 15:22:22 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:22:22Z" level=info msg="gateway: sending udp packet to gateway" addr=10.0.2.2:42565 protocol_version=2 type=PushACK

sudo systemctl status lora-gateway-bridge
● lora-gateway-bridge.service - LoRa Gateway Bridge
   Loaded: loaded (/lib/systemd/system/lora-gateway-bridge.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-08-10 14:51:42 UTC; 37min ago
     Docs: https://docs.loraserver.io
 Main PID: 1283 (lora-gateway-br)
    Tasks: 8
   Memory: 13.4M
      CPU: 759ms
   CGroup: /system.slice/lora-gateway-bridge.service
           └─1283 /usr/bin/lora-gateway-bridge

Aug 10 15:28:46 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:28:46Z" level=info msg="gateway: received udp packet from gateway" addr=10.0.2.2:51990 protocol_version=2 t
Aug 10 15:28:46 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:28:46Z" level=info msg="gateway: sending udp packet to gateway" addr=10.0.2.2:51990 protocol_version=2 type
Aug 10 15:28:52 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:28:52Z" level=info msg="gateway: received udp packet from gateway" addr=10.0.2.2:42565 protocol_version=2 t
Aug 10 15:28:52 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:28:52Z" level=info msg="gateway: stat packet received" addr=10.0.2.2:42565 mac=7276ff00080801db
Aug 10 15:28:52 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:28:52Z" level=info msg="backend: publishing packet" topic="gateway/7276ff00080801db/stats"
Aug 10 15:28:52 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:28:52Z" level=info msg="gateway: sending udp packet to gateway" addr=10.0.2.2:42565 protocol_version=2 type
Aug 10 15:28:56 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:28:56Z" level=info msg="gateway: received udp packet from gateway" addr=10.0.2.2:51990 protocol_version=2 t
Aug 10 15:28:56 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:28:56Z" level=info msg="gateway: sending udp packet to gateway" addr=10.0.2.2:51990 protocol_version=2 type
Aug 10 15:29:06 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:29:06Z" level=info msg="gateway: received udp packet from gateway" addr=10.0.2.2:51990 protocol_version=2 t
Aug 10 15:29:06 ubuntu-xenial lora-gateway-bridge[1283]: time="2017-08-10T15:29:06Z" level=info msg="gateway: sending udp packet to gateway" addr=10.0.2.2:51990 protocol_version=2 type
lines 1-21/21 (END)

RogerioCassares, 2017-08-10:
Hi Brocaar, I have accessed the virtual machine using ssh, and then the directory 

"/home/ubuntu/go/src/github.com/brocaar/lora-app-server/"

I have run a test using the "make test" command, and then it returned for me this bellow: 

Generating combined Swagger JSON
Generating static files
cmd/lora-app-server/main.go:1: running "go-bindata": exec: "go-bindata": executable file not found in $PATH
Makefile:46: recipe for target 'internal/statics' failed
make: *** [internal/statics] Error 1

Do you think it can be responsable for the error? 
Thanks so much for the help!
I looking forward for the positive results!

brocaar, 2017-08-10:
This is not a bug 😉 This error is raised when LoRa App Server can't communicate with the LoRa Server instance. Please check your setup to see if all services are up and running / if there are any errors in the logs.

For further support about your setup, please refer to our forum which is a better place to discuss these things: https://forum.loraserver.io

closed 2017-08-10 by brocaar