brocaar/chirpstack-application-server #134: npm WARN unmet dependency

y0913011011, 2017-08-21:
### Is this a bug or a feature request?
no
### What did you expect?
i want to solve it,but i dnt know how
### What happened?
lchw@vm:lora-app-server$ make requirements ui-requirements
Installing development tools
Installing UI requirements
npm WARN unmet dependency /home/lchw/gowork/src/github.com/brocaar/lora-app-server/ui/node_modules/react-scripts requires whatwg-fetch@'2.0.2' but will load
npm WARN unmet dependency /home/lchw/gowork/src/github.com/brocaar/lora-app-server/ui/node_modules/whatwg-fetch,
npm WARN unmet dependency which is version 1.1.1
npm WARN unmet dependency /home/lchw/gowork/src/github.com/brocaar/lora-app-server/ui/node_modules/react-router/node_modules/history requires warning@'^2.0.0' but will load
npm WARN unmet dependency /home/lchw/gowork/src/github.com/brocaar/lora-app-server/ui/node_modules/react-router/node_modules/warning,
npm WARN unmet dependency which is version 3.0.0

### What version are your using?
go version go1.9rc2 linux/386
### How can your issue be reproduced?

### Could you share your log output?

brocaar, 2017-08-21:
Is this a warning or an error?

y0913011011, 2017-08-21:
warning!but make requirements ui-requirements not works

closed 2017-08-22 by brocaar