brocaar/chirpstack-application-server #135: how can i make requirements ui-requirements sucess?

y0913011011, 2017-08-21:
### Is this a bug or a feature request?
No
### What did you expect?
Compiling Needs package golang.org/x/tools/go/gcexportdata,but i cant access this url.some suggestion let me download from https://github.com/golang/tools.git on the desk,then mv package to golang.org/x/tools/go/gcexportdata.
how can i make requirements ui-requirements sucess?
### What happened?
lchw@vm:lora-app-server$ make requirements ui-requirements
Installing development tools
package golang.org/x/tools/go/gcexportdata: golang.org/x/tools is a custom import path for https://go.googlesource.com/tools, but /home/lchw/gowork/src/golang.org/x/tools is checked out from https://github.com/golang/tools.git
make: *** [requirements] 错误 1

### What version are your using?
go version go1.9rc2 linux/386
### How can your issue be reproduced?

### Could you share your log output?

brocaar, 2017-08-21:
It seems like there is an issue with your Go installation and / or GOPATH. Make sure you have cloned the project into `$GOAPTH/src/github.com/brocaar/lora-app-server`.

y0913011011, 2017-08-21:
my go env:
#go
export PATH=$PATH:/usr/local/go/bin
export GOPATH=/home/lchw/gowork
export GOROOT=/usr/local/go
and clone project into $GOAPTH/src/github.com/brocaar/lora-app-server indeed

y0913011011, 2017-08-21:
when i installed npm,other errores occur!
9793 error code EINTEGRITY
9794 error sha512-JK2bT5tZnLQlNuLa8pjOvUC/c/t4MfXBPZyrNK3C1BBxfX1rMHxFC2DvVCNd8esRjDtEddnwSTlv54sC5gvtMQ== integrity checksum failed when using sha512: wanted sha512-JK2bT5tZnLQlNuLa8pjOvUC/c/t4MfXBPZyrNK3C1BBxfX1rMHxFC2DvVCNd8esRjDtEddnwSTlv54sC5gvtMQ== but got sha512-CoWSz/RtkAd1dAZv9CnR/Kwa0OD9G1/6MdG4bPprmkbieZwT6+s6zGRe5tPpRX+ezGeJIPcLNqz9xK7J8tMnzw==. (138485 bytes)

y0913011011, 2017-08-21:
Cheers,After updated npm and nodejs,it pass

closed 2017-08-21 by y0913011011