brocaar/chirpstack-application-server #136: Possibility to deactivate Node without deleting it

kdannen, 2017-08-21:
### Feature Request
### For testing purposes it is sometimes useful to ignore a node/device. Currently the only possibility to achieve this is by deleting the node. But this means I have to create it completely from scratch when I want to use it again. A small "deactivate" or "pause" button would solve this.

bhatanku1, 2018-10-01:
Hey @brocaar ! Could you please let us know if this feature is planned any time soon. I am also currently deleting the device when I need to deactivate it. 

Thanks a lot.
Regards
Ankur

brocaar, 2018-10-03:
It has been implemented. One can now simply call `DELETE /api/devices/[DevEUI]/activation` to delete the device-activation (not the device) :)

This will be included in the next release.

closed 2018-10-03 by brocaar

bhatanku1, 2018-10-04:
That's awesome @brocaar :) . Thank you very much. Do you already have an estimated date for the next release? 
Regards
Ankur

brocaar, 2018-10-05:
Probably next week or the week after.

bhatanku1, 2018-10-05:
Great! Thanks a ton!

commit 378a96af (2018-10-03): Implement device de-activation API.

Closes #136.