brocaar/chirpstack-application-server #137: Data send in downlink is different on portal

Ankitjd18, 2017-08-21:
### Is this a bug or a feature request?
Query - I sent Base64 encoded data using mqtt downlink, but received some other text in payload on portal
### What did you expect?
Same data to be shown on portal
### What happened?
Different text on portal
### What version are your using?
0.10.0
### How can your issue be reproduced?
Not clear, as this can be something at my end
### Could you share your log output?
NA

brocaar, 2017-08-22:
Please note that sending base64 encoded data does not mean that your node will also use base64 encoding. The base64 encoding is only used as "presentation" in the JSON MQTT interface. In the end the data is transmitted as part of the PHYPayload (bytes) and presented by your node in whatever encoding the node is using. E.g. this could be HEX.

ehsanmqn, 2017-08-22:
Dear brocaar, regarding the @Ankitjd18 problem I am also encountered with this issue. I already have registered a parking sensor using the application server. In the portal, in frame log section I got "zRatGwf/rVWihMptEA3ADVOZWpT8WA==" as _phyPayload_. I also subscribed into the node channel and got "+gABCRaqABAQAQEAARP///////9PTw==" as _payload_. Can you describe me please what this stuffs means and how I am able to decode them? (I thought they are base64 encoding)

brocaar, 2017-08-22:
Would you mind creating a topic at https://forum.loraserver.io/c/lora-app-server? I'm happy to answer your question but the forum is a better place for these kind of questions :-) That way it will not only help you now, but others too in the future!

Ankitjd18, 2017-08-22:
Yeah sure, I have created a topic there with the same subject.

closed 2017-09-04 by brocaar