brocaar/chirpstack-application-server #138: Receiving duplicate message i.e. same fCnt

Ankitjd18, 2017-08-21:
### Is this a bug or a feature request?
Bug - Subscribed to MQTT topic to receive data for all nodes, also receiving data with same fCnt which should be filtered, I guess.
### What did you expect?
Filtered data, single data for unique fCnt.
### What happened?

### What version are your using?
0.10.0
### How can your issue be reproduced?

### Could you share your log output?

brocaar, 2017-08-22:
LoRa Server already does the `FCnt` filtering. Do you have relax frame-counter mode enabled?

Ankitjd18, 2017-08-22:
I have tried with relax frame counter enabled and disabled both. Which mode is it suppose to work? I will try that again.

brocaar, 2017-08-22:
Try without the relaxt frame-counter. Don't forget to reactivate your node again (OTAA or ABP) after making this change.

Ankitjd18, 2017-08-22:
I'll try this.
So duplicates should not come when subscribed to mqtt topic. Is duplicate expected in frame logs on the portal?

brocaar, 2017-08-22:
Duplicated FCnt usage will be logged as invalid MIC or FCnt log-line in LoRa Server.

Ankitjd18, 2017-08-22:
ok thanks

Ankitjd18, 2017-08-22:
Getting the same behaviour with relax frame counter enabled and disabled both. Still receiving duplicate fCnt.

brocaar, 2017-08-22:
Are you using OTAA and did you re-activate your node? In case of ABP, please re-activate the node in the web-interface.

Ankitjd18, 2017-08-22:
Yeah it worked, by re-activating the node (setting fcnt up and down to 0) but from next reset of node, no data is coming. I guess with each reset of the node, we need to reactivate the node using web interface, which does not seem good. Please suggest.

brocaar, 2017-08-22:
A FCnt must always increment, so when you reset your node you either have to store the FCnt values before reset or re-activate (OTAA) after reset. Alternatively you need to use the relax frame-counter mode, but that will pass duplicated frames. Unfortuantely, you can't have all 😉

Ankitjd18, 2017-08-23:
Storing current Fcnt at device end doesn't seem a feasible solution as no of read/write cycle from EEPROM is limited. And I think, this is a good feature to have, as we don't know when the device might reset in the field due to power reset or any other reason.
Can we have any work around? Are you planning to have it in future?

brocaar, 2017-08-23:
Until LoRaWAN 1.1 the only workaround is the relaxt frame-counter mode **when using ABP**, as nothing in the protocol facilitates this to deal with this in a safe way. You either ignore the frame-counters (with the risk of replay attacks) or you respect them. There is no safe way to detect from the network side that your node was reset from the network-server side.

A better way is to use OTAA and send a new join-request after the node was reset. That way all counters will be set to `0` at the network-server side.

Ankitjd18, 2017-08-23:
Ok, we will use OTAA. But, what if join request is missed or does not reach to the server. In this case, what will happen?

brocaar, 2017-08-23:
You are able to detect this at the MAC layer of the node (e.g. it will return a rejected status) and you'll need to try again.

Ankitjd18, 2017-08-23:
Thanks for all this help. I think we can work with this.

closed 2017-08-23 by Ankitjd18

darkfader, 2020-04-03:
"LoRa Server already does the FCnt filtering".
Which part? I'm still seeing this issue via my HTTP integration in ChirpStack 3.9.