brocaar/chirpstack-application-server #142: Pagination in node frame logs

grendy, 2017-08-23:
Hi, how to minimize pagination in node frame logs?
https://forum.loraserver.io/t/pagination-in-node-frame-logs/69

iegomez, 2017-08-25:
@grendy @brocaar  
If you could wait till next week, I'll tackle this. I was thinking about a general solution (not only for frame logs) with a new component that allows to choose a per_page value from a select list (or maybe let the user give a specific value) that'll be rendered in every page that includes pagination. Let me know if that sounds ok to work on it next week.

brocaar, 2017-08-25:
Don't make it too complex :-) Please note that this component receives the number of pages + the current active page. Lets start from there as else it would involve a lot of other code changes. 

Maybe the result for a large set of pages could look like (for different active pages):

[1] [2] [3] [4] [**5**] [6] [7] [8] ... [100]

[1] ... [3] [4] [5] [**6**] [7] [8] [9] ... [100]

[**1**] [2] [3] [4] ... [100]

[1] [**2**] [3] [4] [5] ... [100]

[1] ... [97] [98] [99] [**100**]

[1] ... [96] [97] [98] [**99**] [100]

iegomez, 2017-08-25:
That sounds like a nice and quick solution and I'd be glad to implement it. But I'd also like to mention another way to deal with big amounts of data, which I implemented for nodes saved records and other components, which is allowing to filter by date and time. For example, here is a screenshot of what my saved data view looks like:

[image]

In most cases I prefer the filter approach over the pagination one, specially when it is data related, because date and time are usually relevant data-wise. The drawback is that the storage and api need a little change to handle start and end parameters, so it goes beyond a little modification to the frontend only. 

What do you think about this approach?

brocaar, 2017-08-26:
I like it, but that would require a lot of more work (as it requires API changes too) and still would not solve the original issue. The filtered query could potentially still return 100 pages. So I think a better pagination regardless of filtering would still be needed.

I see the following responsibilities:

- The component performing the data query should be responsible for applying any filters. As this component is tied to the datamodel, it knows about which potential filters could be applied. Based on the resultset it renders the current page and knows about the number of pages available.

- The pagination component (the component we're discussing) only knows about the total number of pages + current page and knows the best way to render this. That way it is never tied to the datamodel for which it is paginating.

Note that the pagination component is also used for users, organizations, applications, nodes etc. The solution that I propose would therefore would also directly solve a large number of pages for these components. E.g. this is how it is being used:

https://github.com/brocaar/lora-app-server/blob/master/ui/src/views/users/ListUsers.js#L87
https://github.com/brocaar/lora-app-server/blob/master/ui/src/views/gateways/ListGateways.js#L140

iegomez, 2017-08-26:
Well yes, they are indeed two different components, and I agree that the pagination one doesn't need to know about the model. But then again, the ability to paginate is tied to the API allowing offset and limit parameters and the model using them to query for results, which is spread all over the application (making the pagination component uniform along the frontend). So is date, as every record has a created_at attribute, so it is also natural to allow to query for specific dates or range of dates.

Maybe they should be treated as 2 separate enhancements, as they solve different problems. 

Anyway, I'll follow your recommendation for the pagination rendering (using ... and limited number of pages) and implement it next week, and maybe in the future a common date filtering component could be included too.

I'll let you know when it is done.

iegomez, 2017-08-29:
@brocaar I took some time today and got the pagination working as you intended, just with two little "add-ons":

1) The default "steps" shown backward and forward are 3, but this may also be set through props with "pageSteps".
2) The gap is also a link pointing to the center page in the gap. For instance, if your pagination looks like [1] [2] [3] [4] [...] [100], then the [...] link points to page 52. This allows to binary jump and get to pages quicker when there are too many.

Again, you can check it at https://github.com/iegomez/lora-app-server. It contains the changes to Pagination.js along with some past changes for the node search feature requested in [#120](https://github.com/brocaar/lora-app-server/issues/120). If you are ok with the implementations, let me know to create a PR.

brocaar, 2017-09-04:
@iegomez just tested your pagination solution and I think this improves the navigation a  lot :-) I've cherry-picked your commit into the master branch, see https://github.com/brocaar/lora-app-server/commit/3040a051ca3e5544842df405e565f4e9df197e0c.

Regarding the search solution I need to do some more research around performance (e.g. what changes this would require on the DevEUI column in order to avoid table scans).

closed 2017-09-04 by brocaar

grendy, 2017-09-04:
Thanks @iegomez @brocaar sorry for late response