brocaar/chirpstack-application-server #145: desc = maximum payload size exceeded

interfaith, 2017-08-29:
When using Mdot an AT-COMMAND can easily allow larger than 12 bytes to come from the Mdot node.
However when sending TO the Mdot the lora-app-server log shows the above error.

Maybe the Mdot will accept it BUT the lora-app-server rejects sending more than 12 bytes.?

The JSON file has SF set to 7 , for maximum datarate according to the semtech datasheet.

brocaar, 2017-08-30:
Could you please share any logs? Especially the uplink JSON message posted to the `application/+/node/+/rx` topic would be really helpful.

Also:
- which ISM band are you using?
- which receive window did you setup for your node?

interfaith, 2017-08-30:
class c
usa

Aug 29 10:25:21  lora-app-server[1463]: time="2017-08-29T10:25:21-07:00"
level=error msg="handle data-down payload error: push data-down error: rpc
error: code = InvalidArgument desc = maximum payload size exceeded"
application_id=2 dev_eui=008000000000eb0a

interfaith, 2017-08-30:
MDOT config
[TRACE] RX2 on freq: 923300000
[TRACE] RX DR: 8 SF: 12 BW: 2 CR: 1 PL: 8 STO: 12 CRC: 0 IQ: 1
[TRACE] Stats: Up: 0 Down: 0 DupTx: 0 CRC Errors: 0
[INFO] mDot Ready
[INFO] mbed-os library version: 144
[INFO] defaulting Dot configuration
[INFO] =====================
[INFO] general configuration
[INFO] =====================
[INFO] version ------------------ 3.0.0-mbed144
[INFO] device ID/EUI ------------ 008000000000eb0a
[INFO] default channel plan ----- US915
[INFO] current channel plan ----- US915
[INFO] frequency sub band ------- 1
[INFO] public network ----------- on
[INFO] =========================
[INFO] credentials configuration
[INFO] =========================
[INFO] device class ------------- C
[INFO] network join mode -------- OTA
[INFO] network name -------------
[INFO] network phrase -----------
[INFO] network EUI -------------- 2888f370eabba0b0
[INFO] network KEY -------------- b08c1b98fd48c0ec29c749b2027d22bd
[INFO] ========================
[INFO] communication parameters
[INFO] ========================
[INFO] acks --------------------- off, 0 attempts
[INFO] TX datarate -------------- DR0
[INFO] TX power ----------------- 30 dBm
[INFO] antenna gain ------------- 3 dBm
[INFO] LBT ---------------------- off
[INFO] changing network mode to class C
[INFO] Set radio to Public Mode
[WARNING] Radio Busy, cannot open Rx Window
[INFO] saving configuration
[INFO] =====================
[INFO] general configuration
[INFO] =====================
[INFO] version ------------------ 3.0.0-mbed144
[INFO] device ID/EUI ------------ 008000000000eb0a
[INFO] default channel plan ----- US915
[INFO] current channel plan ----- US915
[INFO] frequency sub band ------- 1
[INFO] public network ----------- on
[INFO] =========================
[INFO] credentials configuration
[INFO] =========================
[INFO] device class ------------- C
[INFO] network join mode -------- OTA
[INFO] network name -------------
[INFO] network phrase -----------
[INFO] network EUI -------------- 2888f370eabba0b0
[INFO] network KEY -------------- b08c1b98fd48c0ec29c749b2027d22bd
[INFO] ========================
[INFO] communication parameters
[INFO] ========================
[INFO] acks --------------------- off, 0 attempts
[INFO] TX datarate -------------- DR3
[INFO] TX power ----------------- 20 dBm
[INFO] antenna gain ------------- 3 dBm
[INFO] LBT ---------------------- off
[INFO] waiting to join

and from the server packet forwarder global_conf.json SF8 was changed to 7
to try for more speed larger packet

  "chan_Lora_std": {
            /* Lora MAC channel, 500kHz, SF8, 903.0 MHz */
            "enable": true,
            "radio": 0,
            "if": 300000,
            "bandwidth": 500000,
            "spread_factor": 7
        },

brocaar, 2017-08-30:
Please note that Class-C uses the RX2 receive window. Make sure this is set correctly in LoRa App Server (under *Advanced network settings* tab of your node). After changing this value re-activate your node (OTAA) as the RX2DR is sent to your node as part of the join-accept frame.

interfaith, 2017-08-30:
The *Advanced network settings* tab also has "RX2 datarate" must that be
set to something ?

brocaar, 2017-08-30:
To the data-rate you wish to use for the RX2 receive-window (used by Class-C). Please refer to the LoRaWAN Regional Parameters for valid data-rates. By default this is set to DR `0`, meaning for your band you can send up to `11` bytes explaining your issue.

As mentioned before, don't forget to re-activate (OTAA) your node after changing this value, as this is communicated to your node as part of the join-accept frame.

interfaith, 2017-08-30:
SO instead of 0 then 5 would mean DR5 ? Maybe there is a document that
details 'advanced settings' ?

I will be sure to let you know if the changes work. So the lora-app-server
"regulates" the amount /size of data that can be sent ?

brocaar, 2017-08-30:
Yes, 5 means DR5 for (RX2). LoRa App Server only keeps the configuration for the RX2 data-rate and passes this to the node on OTAA (so that the node knows which data-rate to use when listening for RX2).

interfaith, 2017-08-30:
yeah !! the 'advanced settings' fix did the trick . Thanks for all the
hints !

closed 2017-09-04 by brocaar