brocaar/chirpstack-application-server #147: REST API Application Count stuck at 10, repeating appID number one.

PhatHub, 2017-08-31:
### Is this a bug or a feature request?
Bug

### What did you expect?
I was expecting Application count of 1 when making a RESTful GET request.
Since there is only one Application that exists currently, I was expecting that application to show only once, regardless of the number in the limit field.

### What happened?
Regardless of the number in the limit field, I get an Application count of 10.
If I set a limit from any value of 1 to 10, the fields for application with the id of "1" appears that many times.
If I set a limit over the number ten, the fields for application with the id of "1" only appears ten times.

### What version are your using?
Version: 0.10.0
Commit: 621f55674f72d1c947fe9866a3b85f8bfc5aa40c

### How can your issue be reproduced?
Example REST command (I set the limit to 5 so it won't be too long): 
`curl -X GET --header 'Accept: application/json' --header 'Grpc-Metadata-Authorization: [token]' 'https://[server]/api/applications?limit=5&offset=1'`

The output I get is:
> {
>   "totalCount": "10",
>   "result": [
>     {
>       "id": "1",
>       "name": "P1ac3-H01d3r",
>       "description": "This is a PLACE HOLDER.",
>       "rxDelay": 1,
>       "rx1DROffset": 0,
>       "rxWindow": "RX2",
>       "rx2DR": 0,
>       "relaxFCnt": true,
>       "adrInterval": 0,
>       "installationMargin": 0,
>       "isABP": true,
>       "isClassC": false,
>       "organizationID": "1"
>     },
>     {
>       "id": "1",
>       "name": "P1ac3-H01d3r",
>       "description": "This is a PLACE HOLDER.",
>       "rxDelay": 1,
>       "rx1DROffset": 0,
>       "rxWindow": "RX2",
>       "rx2DR": 0,
>       "relaxFCnt": true,
>       "adrInterval": 0,
>       "installationMargin": 0,
>       "isABP": true,
>       "isClassC": false,
>       "organizationID": "1"
>     },
>     {
>       "id": "1",
>       "name": "P1ac3-H01d3r",
>       "description": "This is a PLACE HOLDER.",
>       "rxDelay": 1,
>       "rx1DROffset": 0,
>       "rxWindow": "RX2",
>       "rx2DR": 0,
>       "relaxFCnt": true,
>       "adrInterval": 0,
>       "installationMargin": 0,
>       "isABP": true,
>       "isClassC": false,
>       "organizationID": "1"
>     },
>     {
>       "id": "1",
>       "name": "P1ac3-H01d3r",
>       "description": "This is a PLACE HOLDER.",
>       "rxDelay": 1,
>       "rx1DROffset": 0,
>       "rxWindow": "RX2",
>       "rx2DR": 0,
>       "relaxFCnt": true,
>       "adrInterval": 0,
>       "installationMargin": 0,
>       "isABP": true,
>       "isClassC": false,
>       "organizationID": "1"
>     },
>     {
>       "id": "1",
>       "name": "P1ac3-H01d3r",
>       "description": "This is a PLACE HOLDER.",
>       "rxDelay": 1,
>       "rx1DROffset": 0,
>       "rxWindow": "RX2",
>       "rx2DR": 0,
>       "relaxFCnt": true,
>       "adrInterval": 0,
>       "installationMargin": 0,
>       "isABP": true,
>       "isClassC": false,
>       "organizationID": "1"
>     }
>   ]
> }
> 

### Could you share your log output?
No log file, and the journalctl log only has `handler/mqtt` and `data-down` data.

brocaar, 2017-09-01:
Thanks for reporting! I think I found the issue. Probably your user is assigned to both the organization and to the application. This should of course not happen and I will fix it, but it explains why this hasn't been reported yet. 

To be sure, could you make a database dump of the LoRa App Server database and send this by e-mail (info at brocaar dot com) so I can test my fix against your dataset? (you could remove all nodes from this set / blank out all app keys).

Example command:

```
pg_dump -h localhost -U loraserver_as -W loraserver_as > loraserver_as.sql
```

PhatHub, 2017-09-01:
Ok! I've emailed you. I hope it's helpful!

commit a484586a (2017-09-04): Fix duplicated resultset.

Closes #147.
closed 2017-09-04 by brocaar