brocaar/chirpstack-application-server #151: message latency in class c

interfaith, 2017-09-11:
Can anything be adjusted to minimize latency. 
Is there a latency with the loraserver.io? when queuing uploads perhaps?

brocaar, 2017-09-14:
Could you be a bit more specific about the issue you're facing and what kind of latency you mean? E.g. is this in ms, seconds, ..?

interfaith, 2017-09-14:
to do 2400 baud handshake. I guess about 10msec could be the max latency.

brocaar, 2017-09-19:
> to do 2400 baud handshake. I guess about 10msec could be the max latency.

Sorry, I don't really understand the issue. What kind of handshake do you mean? Please note that the "latency" can be broken up in two parts:

- "Latency" introduced by LoRa App Server + LoRa Server
- "Latency" introduced by the gateway and transmission

Please note that LoRaWAN is not a fast modulation. Depending the data-rate, it takes way more than 10ms to transmit a LoRaWAN frame. See also: https://www.thethingsnetwork.org/forum/t/spreadsheet-for-lora-airtime-calculation/1190.

interfaith, 2017-09-27:
> lora-airtime-calculation/1190.

Now, turn around times are as low as 500msec/class-c, which is workable in
this case.
However constraints such as duty-cycle, may also hamper the need to do a
few seconds of handshaking once a day. Listen before talk LBT is one option
that may help.
I hope all the lora rules will allow a few seconds of handshaking once a
day :) ha ha

closed 2017-11-19 by brocaar