brocaar/chirpstack-application-server #153: HTTP integration not working

Ankitjd18, 2017-09-14:
### Is this a bug or a feature request?
Bug/Doubt - We tried HTTP integration to receive the data  but receiving blank in request body
### What did you expect?
Data in same json structure as in documentation
### What happened?
Received blank data in request body
### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?
Are we missing something? Please let us know what are we missing.

brocaar, 2017-09-15:
Could you create a https://requestb.in endpoint, use this for the HTTP integration and share the request bin URL?

Ankitjd18, 2017-09-16:
here is the url - https://requestb.in/1jgeu6a1?inspect

Ankitjd18, 2017-09-16:
I think I am missing the content-type of the request body. Can you let me know what is it? I was under the impression of json req body from the docs.

C1AFFA, 2017-09-16:
Hi Ankitjd18,
I had the same problem. Probably you have to handle the request raw body. I solved using file_get_contents("php://input"); in my php handler.
Hope this can be helpful!

Ankitjd18, 2017-09-16:
Thanks, I am working on that too.

Ankitjd18, 2017-09-16:
Is the request body encoded with gzip? I am trying to get this in my node handler, but unable to figure out a working solution to get this request data. If you can tell me how is the request body encoded and which format the data is in, it might be of some help.

C1AFFA, 2017-09-16:
I solved doing this:

`$req_dump = file_get_contents("php://input");
$jObj = json_decode($req_dump);
$data =  $jObj ->data;
$fp = fopen('uplink.log', 'a');
fwrite($fp, $data);
fclose($fp);`

The request body is a simple json, so decoding it you obtain an Object. In my case I wanted to pull out the field 'data' and store it in a file. I guess you wanna do the same :D

brocaar, 2017-09-18:
Whoops, you're right! The content-type is missing. Will fix that :-)

Ankitjd18, 2017-09-18:
Ok thanks.

Ankitjd18, 2017-10-02:
@brocaar Did you get any chance to fix this?

brocaar, 2017-10-03:
Yes and it will be released today or tomorrow together with an other new feature :-) Please note that you can set your own headers, so for the time being you could set the content-type to application/json yourself too.

commit d656c686 (2017-10-03): Set default Content-Type header.

Closes #153.
closed 2017-10-03 by brocaar