brocaar/chirpstack-application-server #158: I can send down data through the API.How do I use API interfaces to see the payload of the node's uplink data?Are we missing something? Please let us know what are we missing.

SnailQuick, 2017-10-02:
### Is this a bug or a feature request?

### What did you expect?

### What happened?

### What version are your using?

### How can your issue be reproduced?

### Could you share your log output?

brocaar, 2017-10-03:
There is no API method for uplink data. Please see: https://docs.loraserver.io/lora-app-server/integrate/data/

closed 2017-10-03 by brocaar

SnailQuick, 2017-10-06:
very thanks！

SnailQuick, 2017-10-06:
API -> Node  ->
[image]
get information :   " "phyPayloadJSON": "{\"mhdr\":{\"mType\":\"ConfirmedDataDown\",\"major\":\"LoRaWANR1\"},\"macPayload\":{\"fhdr\":{\"devAddr\":\"07d1dbda\",\"fCtrl\":{\"adr\":false,\"adrAckReq\":false,\"ack\":true,\"fPending\":false},\"fCnt\":6,\"fOpts\":null},\"fPort\":44,\"frmPayload\":[{\"bytes\":\"nsRouyuy\"}]},\"mic\":\"936437a8\"}"”

"nsRouyuy" What's the data?