brocaar/chirpstack-application-server #160: Feature request:  Display gateway bridge, server and app server versions

sslupsky, 2017-10-04:
### Is this a bug or a feature request?
Feature request

### What did you expect?
Would it be possible to have a page in the App Server that displayed the current versions of the installed software for the Gateway Bridge, Server and App Server?

brocaar, 2017-10-05:
I don't think that would work. LoRa App Server should not have access to LoRa Gateway Bridge as this sits in-between the gateway (or on the gateway) and LoRa Server. Also, LoRa App Server could be connected to one LoRa Server instance part of a cluster. So the information you would see would only reflect a single instance, not your complete deployment.

brocaar, 2019-06-24:
See #316

closed 2019-06-24 by brocaar