brocaar/chirpstack-application-server #161: New gateways are invisible in lora-app-server

sunblade, 2017-10-05:
I have running lora-app-server version 0.13.0 on Linux x64

Gateway is not visible in lora-app-server, but works okay.

```
loraserver_ns=# SELECT * from gateway WHERE mac = '\x00000000000010ab';
 mac | name | description | created_at | updated_at | first_seen_at | last_seen_at | location | altitude | channel_configuration_id
-----+------+-------------+------------+------------+---------------+--------------+----------+----------+--------------------------
(0 rows)
```
```
loraserver_ns=# SELECT * from gateway WHERE mac = '\x00000000000010ab';
        mac         |       name       | description |          created_at           |          updated_at           |        first_seen_at         |         last_seen_at          |      location      | altitude | channel_configuration_id
--------------------+------------------+-------------+-------------------------------+-------------------------------+------------------------------+-------------------------------+--------------------+----------+--------------------------
 \x00000000000010ab | 00000000000010ab |             | 2017-10-05 18:52:38.061582+02 | 2017-10-05 19:04:08.088423+02 | 2017-10-05 18:52:38.06157+02 | 2017-10-05 19:04:08.088418+02 | (49.73985,18.6351) |      320 |
(1 row)
```
After adding this GW manually, app-server is showing GW with all data (also stats from the past)

Log from loraserver:
```
Oct 05 19:06:08 lora-srv1 loraserver[28251]: time="2017-10-05T19:06:08+02:00" level=info msg="backend/gateway: gateway stats packet received" mac=00000000000010ab
```

Before I was using very old loraserver and app-server (with one PSQL DB). Currently I am running two databases (NS and AS)

Also, when trying to edit some of existing GW via lora-app-server, I am able to see errors:
```
Error object does not exist (code: 5)
Error gateway does not exist (code: 5)
```

Debug from PostgreSQL
```
2017-10-05 19:10:58.173 CEST [28258] [unknown] loraserver_ns@loraserver_ns LOG:  execute <unnamed>:
			update gateway set
				name = $2,
				description = $3,
				updated_at = $4,
				first_seen_at = $5,
				last_seen_at = $6,
				location = $7,
				altitude = $8,
				channel_configuration_id = $9
			where mac = $1

2017-10-05 19:10:58.173 CEST [28258] [unknown] loraserver_ns@loraserver_ns DETAIL:  
			parameters: $1 = '\x00000000000000ab', $2 = 'TEST_GW_NAME', $3 = 'TEST_Descr', $4 = '2017-10-05

2017-10-05 19:10:58.174 CEST [28239] [unknown] loraserver_as@loraserver_as LOG:  statement: COMMIT
```

brocaar, 2017-10-06:
LoRa Server can create gateways automatically (to track stats), but LoRa App Server cannot, as it needs to know which gateway belongs to which organisation. You need to use the create gateway form from the UI (or the API endpoints) to create gateways. For further support questions, please use https://forum.loraserver.io/

closed 2017-10-06 by brocaar

brocaar, 2017-10-06:
Yes that could be, but I think at that point there were no organizations.