brocaar/chirpstack-application-server #163: Error installer

duracellkonijn22, 2017-10-10:
### Is this a bug or a feature request?
Kind of bug

### What did you expect?
Something working :-)

### What happened?
sudo echo "deb https://repos.loraserver.io/${DISTRIB_ID,,} ${DISTRIB_CODENAME} testing" | sudo tee /etc/apt/sources.list.d/loraserver.list

this line doesn't get all packages correctly (at the end).

Tomorrow I try with Ubuntu or other version of Debian

### What version are your using?
Debian Stretch

### How can your issue be reproduced?
Just follow the tutorial until that line

### Could you share your log output?
Just closed the virtual machine, but some kind of packages are reachable, some not (in the end)

duracellkonijn22, 2017-10-12:
Idem with Ubuntu Jessie. Some kinds of packets from loraserver not found

brocaar, 2017-10-12:
> Idem with Ubuntu Jessie. Some kinds of packets from loraserver not found

I suppose you mean Debian Jessie?

> some kind of packages are reachable, some not (in the end)

Please share the actual log message as I can't reproduce it (both with Jessie as with Stretch). Without any log messages there is not much I can do. Could you share the output of the following commands:

* `sudo apt-get update`
* `sudo apt-get install lora-gateway-bridge loraserver lora-app-server`

closed 2017-10-22 by brocaar