brocaar/chirpstack-application-server #164: Maximum Payload Size Exceeded Error after updating to Lora-app-server version 0.13.1

maxximal, 2017-10-13:
Hello, currently I'm having some problem on sending more than 65 bytes of data from app-server to end-device. According to LoRaWAN specification v.1.0.2 Regional Parameters KR Band, the maximum payload size with SF7 should be 242 bytes. This configuration works with previous version of LoRa-App-Server. However, I got error after lora-app-server version is updated to 0.13.1. I didn't change any configuration that's working before.
### Is this a bug or a feature request?
Node configuration stay the same as the previously working one. Setup:
Class C; Receive window RX2; RX2 data-rate = 5 (SF7); Freq Band Config: KR920_923;
### What did you expect?
maximum payload size = 242 btyes
### What happened?
{"error": "push data-down error: rpc error: code = InvalidArgument desc = maximum payload size exceeded",
  "code": 2}
when the payload is bigger than 65 bytes.
### What version are your using?
Lora-app-server version 0.13.1
loraserver version 0.21.0
### How can your issue be reproduced?
tested through Lora App Server REST API sending more than 65 byte of packet through Downlink queue

is this a bug or is there any steps that I'm missing?

Thank you.
Eddy.

maxximal, 2017-10-23:
After digging a little bit deeper into the sources, I concludes that I might've put the question in the wrong area (should be in loraserver issue). Currently I'm still having the same problem on sending data into end-device. When I tried to published more than 65 bytes, the same error still occurred. During my debugging process, I found that the CTXDataRate is 0 instead of 5. I don't have any problem on receiving 242 bytes of data from the end-device, with the CTXDataRate is shown correctly at 5, same with the RX2Datarate. Does the loraserver has some limitation on sending data, with maximum payload only 65 bytes?

Thank you.
Eddy.

brocaar, 2017-10-23:
LoRa Server validates the max-payload size based on the data-rate used for transmission. These values are taken from the LoRaWAN Regional Parameters document and are ISM band specific. 

You mention that the data-rate is `0` is this because your node was configured this way (in LoRa App Server application / node settings)?

maxximal, 2017-10-23:
Here is my Application/Node settings:
{
  "id": "1",
  "name": "RPI3_KR920",
  "description": "LoRaServer with Raspberry PI3 configuration",
  "rxDelay": 0,
  "rx1DROffset": 0,
  "rxWindow": "RX2",
  "rx2DR": 5,
  "relaxFCnt": false,
  "adrInterval": 0,
  "installationMargin": 0,
  "isABP": false,
  "isClassC": true,
  "organizationID": "1"
}
it worked on previous version before I updated the loraserver and lora-app-server with the current version. So I'm actually confused with the error that I got suddenly after updates.

brocaar, 2017-11-19:
Could you test with the latest release and tell me if the issue is still there? https://forum.loraserver.io/t/release-lora-server-0-22-0-lora-app-server-0-14-x-read-carefully/350

maxximal, 2017-11-20:
Hi.

I have installed the latest release (Loraserver version 0.22.0 & lora-app-server version 0.14.0) and performed some testings with it. The results are pretty much the same. Here are my configurations and my tests results:

`{
      "devEUI": "4786e6ed02001920",
      "name": "RHF3M076",
      "applicationID": "2",
      "description": "LoRa RisingHF with AT-Command",
      "deviceProfileID": "3232ff24-e0fd-4ce7-aaa5-f35761570030",
      "deviceProfileName": "PS01_KR920_Config"
}
{
  "deviceProfile": {
    "deviceProfileID": "3232ff24-e0fd-4ce7-aaa5-f35761570030",
    "supportsClassB": false,
    "classBTimeout": 0,
    "pingSlotPeriod": 0,
    "pingSlotDR": 0,
    "pingSlotFreq": 0,
    "supportsClassC": true,
    "classCTimeout": 0,
    "macVersion": "1.0.2",
    "regParamsRevision": "B",
    "rxDelay1": 1,
    "rxDROffset1": 0,
    "rxDataRate2": 5,
    "rxFreq2": 0,
    "factoryPresetFreqs": [
      0
    ],
    "maxEIRP": 14,
    "maxDutyCycle": 0,
    "supportsJoin": true,
    "rfRegion": "SouthKorea920",
    "supports32bitFCnt": false
  },
  "name": "PS01_KR920_Config",
  "organizationID": "2",
  "networkServerID": "1",
  "createdAt": "2017-11-05T22:16:31.028304-08:00",
  "updatedAt": "2017-11-14T23:28:17.837824-08:00"
}`

Result from MQTT subscribe:
`application/2/node/4786e6ed02001920/rx {"applicationID":"2","applicationName":"PSTag_KR920","deviceName":"RHF3M076","devEUI":"4786e6ed02001920","rxInfo":[{"mac":"b827ebfffe3a4a7a","rssi":-30,"loRaSNR":9.5,"name":"Rxhf_RHF0M301","latitude":36.3610182,"longitude":127.32927569999998,"altitude":0}],"txInfo":{"frequency":922500000,"dataRate":**{"modulation":"LORA","bandwidth":125,"spreadFactor":7},**"adr":false,"codeRate":"4/5"},"fCnt":34,"fPort":8,"data":"MDEyMzQ1Njc4OTAxMjM0NTY3ODkwMTIzNDU2Nzg5MDEyMzQ1Njc4OTAxMjM0NTY3ODkwMTIzNDU2Nzg5MDEyMzQ1Njc4OTAxMjM0NTY3ODkwMTIzNDU2Nzg5MDEyMzQ1Njc4OTAxMjM0NTY3ODkwMTIzNDU2Nzg5MDEyMzQ1Njc4OTAxMjM0NTY3ODkwMTIzNDU2Nzg5MDEyMzQ1Njc4OTAxMjM0NTY3ODkwMTIzNDU2Nzg5MDEyMzQ1Njc4OTAxMjM0NTY3ODkwMTIzNDU2Nzg5MDEyMzQ1Njc4OTAxMjM0NTY3ODkwMTIzNDU2Nzg5MDE="}`

my LoRa end-device was able to send 242 bytes into the loraserver, however the loraserver can only post queue up-to 65 bytes.

[image]
[image]

Additionally, I also found that MQTT publish command will not give any _payload-exceed_ error response, even though I intentionally publish 260 bytes (definitely should be error). It just doesn't appear in DownlinkQueue API and the End-device also receive nothing.
[image]

I will try to investigate more and hopefully this results might be helpful for you.

brocaar, 2017-11-20:
Thanks for testing again. I think I know what your issue is. The DeviceProfile RX2DR is for ABP devices, maybe I should state this more clearly in the documentation of the API. For OTAA devices, the LoRa Server configuration is applied (which I think is better as this is in the domain of the network). See 

```
--rx2-dr value                          rx2 data-rate (when set to -1, the default rx2 data-rate will be used) (default: -1) [$RX2_DR]
```

https://docs.loraserver.io/loraserver/install/config/

maxximal, 2017-11-20:
I think I might have messed up some part of the Device Profile through API. I did set the Device profile as Support Join (OTAA) and Support Class C as you can see in the picture. 
[image]
[image]
However, I changed the rxDelay1 from _"rxDelay1": 0 into "rxDelay1": 1_, in order to comply with LoRaWan Regional Parameter v1.1 (KR920-923 Default Settings) and rxDataRate2  _from "rxDataRate2": 0 into "rxDataRate2": 5_, in order to test the max packet transmission. The original configuration does not effect the _payload exceeded_ problem. I will change back the Device Profile with the original value and re-perform the test once again.

brocaar, 2017-11-20:
Please note that these fields only affect ABP devices. OTAA uses the LoRa Server global config (the config that I mentioned). When you update these, issue a new OTAA join :-)

maxximal, 2017-11-21:
Hi, 
I have retry the tests after re configuring the Device Profile and issuing new OTAA join for my end-devices. In the end, I still got the same _Payload Exceeded_ issue when I try to queue more than 65 bytes from loraserver to end-device. 
[image]
The loraserver itself has no problem on receiving 242 bytes of confirmed message data from my end-device.
[image]

brocaar, 2017-11-21:
Could you post the configuration of the `loraserver` service?

maxximal, 2017-11-21:
Here's my loraserver configuration file.
[loraserver.txt](https://github.com/brocaar/lora-app-server/files/1490663/loraserver.txt)

brocaar, 2017-11-21:
I don't see the

> --rx2-dr value                          rx2 data-rate (when set to -1, the default rx2 data-rate will be used) (default: -1) [$RX2_DR]

Config in your config file.

Please add, `RX2_DR=5` to it, restart `loraserver` and perform a new OTAA activation.

maxximal, 2017-11-21:
Oh, Sorry about that. apparently I missed that one on the file. I did put _RX2_DR=5_ in it tho.
> > # Added config file by Eddy for max payload
> RX2_DR=5
I re-checked the config file and it was in the end of my config file.

brocaar, 2017-11-22:
Thanks for confirming that this is not a config issue. I've finally been able to reproduce the issue. Will follow up soon with a fix :-) Thanks for providing all the information for debugging 👍

closed 2017-11-22 by brocaar

brocaar, 2017-11-22:
I've fixed the issue and this will be released as part of LoRa Server 0.22.1 (binaries will be available in a couple of minutes).

maxximal, 2017-11-23:
Thanks for your hard work! I'll check the latest release today and I'm glad that I can help you on tracing these issues.  👍