brocaar/chirpstack-application-server #166: Downlink in class C nodes also being sent on receiving uplink

Ankitjd18, 2017-10-16:
### What did you expect?
Downlink to be sent immediately as also shows immediately to true in frame logs
### What happened?
Downlink was sent upon receiving the next uplink as in class A nodes
Is there some setting required in order for class C nodes to send downlink properly.

brocaar, 2017-10-17:
Yes, you need to configure the application or node as Class-C device. Also you need to re-activate the node after changing this setting.

Ankitjd18, 2017-10-17:
I have configured the node as Class-C device and also re-activated but it still is behaving like Class-A device.
In frame logs, I can see the `immediately` field to be `true` but still sent on receiving a uplink.

brocaar, 2017-10-17:
To be sure, that is not because your application sends downlink frames on uplink?

Ankitjd18, 2017-10-17:
`immediately` field is set to be `true` for Class-C devices so that downlink can be sent by the server immediately. This field is set to `false` in Class-A devices. This is an observation I made.
I am still not sure of what is causing this issue. I have tried several times, but downlink reaches the device only after it has sent a uplink to the server.

kammyy, 2017-10-18:
@brocaar : Hi, our application is not configured to send downlink upon receiving uplink.

We send the downlink command to node from application and after doing so we can see the downlink listed in frame logs section with immediately tags as "true". But the downlink is sent to node after next uplink.

Ankitjd18, 2017-10-18:
We tried switching between `OTA` and `ABP`. It works fine in case of `OTA` device but not for `ABP`. Is this something to do be the activation?

kammyy, 2017-10-19:
@brocaar : Any comments/suggestions ?

thnx

brocaar, 2017-11-19:
It might be related to your device firmware, this is hard for me to test. Could you also test if your issue persists with the latest release: https://forum.loraserver.io/t/release-lora-server-0-22-0-lora-app-server-0-14-x-read-carefully/350

Ankitjd18, 2017-11-20:
Sure, we will test with the latest release and let you know of our findings.

brocaar, 2017-12-07:
I've just released LoRa App Server 0.15.0 and LoRa Server 0.23.0. This new release has refactored queue logic (LoRa Server is now responsible for the downlink queue). Also you'll get a (n)ACK on confirmed Class-C transmissions.

closed 2017-12-11 by brocaar