brocaar/chirpstack-application-server #168: Confirmed Downlink retry

kammyy, 2017-10-18:
### What happened?
When sending a confirmed downlink to a C type node, when the device does not send "ACK" for the downlink then the server sends the downlink multiple times.

For how many times the server keeps on sending the downlink to node in case the ACK does not come from the node ?

How to change this number of retry for downlink ?

thnx

kammyy, 2017-10-21:
@brocaar : any comments ?

Thanks

brocaar, 2017-11-19:
Currently it will be re-sent until an ACK is received. I'm currently refactoring the downlink queue and it will be possible to give a retry count per downlink message (through MQTT or the rest API) after these changes have been released.

closed 2017-11-19 by brocaar