brocaar/chirpstack-application-server #173: RX2 Datarate

alexanderplanar, 2017-10-23:
### Is this a bug or a feature request?
Bug, I guess, I'm not sure, maybe I do something wrong way.

### What did you expect?
Change RX2 Datarate for downlinks, and send downlinks with datarate **DR_3 (SF9)**

### What happened?
The downlink packet are sent always with default parameters: **DR_0 (SF12 / BW125) on 869.525MHz**

### What version are your using?
$ lora-app-server -h
**0.13.1**

### How can your issue be reproduced?
Use non-standart RX2 window parameters, e.g. **DR_3(SF9 / BW125)** and try to receive the downlinks on the node.

### Could you share your log output?
I have the logs from the packet forwarder on gateway:

JSON down: {"txpk":{"imme":true,"freq":869.525,"rfch":0,"powe":14,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":14,"data":"oFPwwQYAcAABZhOmJTg="}}

I see this record each time I send the downlink using "mosquitto_pub" for my node.
As far as I understand, the packet forwarder receives parameters of the channel the packet must be emitted with.  Even I change the "RX2 data-rate" at "Advanced Network Settings" tab in Application Server Web page, the record remains the same, specifically "SF12BW125". 
I tried the values 0, 1, 2, ... 9, 10, 11, 12.

Moreover, even I change the "RX2 data-rate", I see, that the emitted packet by application server has the following parameters in "Raw frame logs":
```
codeRate:"4/5"
**frequency:"869525000"**
immediately:true
mac:"aa555a0000000101"
power:14
timestamp:0
▶ dataRate:{} 4 keys
    modulation:"LORA"
    bandWidth:125
    **spreadFactor:12**
    bitrate:0
```

P.S. What number I should set, 0, 1, 2... corresponding to DR0, DR1, DR2, or 9, 10, 11, 12... corresponding to SF9, SF10, etc. ?

Thanks in advance!!!

alexanderplanar, 2017-11-17:
Any ideas? It is still actual for me...

brocaar, 2017-11-19:
In the latest release, this setting has been moved to LoRa Server:

```
 --rx2-dr value                          rx2 data-rate (when set to -1, the default rx2 data-rate will be used) (default: -1) [$RX2_DR]
```

https://docs.loraserver.io/loraserver/install/config/

Note that the RX2 data-rate will be sent only on OTAA. For ABP you need to enter the RX2 data-rate in the device-profile.

Please note that it won't be possible anymore to pick the receive-window from the UI. In the future LoRa Server will decide which RX window to use for downlink.

alexanderplanar, 2017-11-20:
**Thanks for your reply!**
Now I use LoRa Server version 0.21.0 and LoRa Application Server 0.13.1.
Does this setting work in my conditions? 
Is it required to update the applications to use this setting?
Can you explain your descision about moving the functionality of choosing the RX-window to LoRa Server? I guess, the ability to choose the RX window from UI is was a good idea, is'nt it?

I use OTAA. As I understand, the device profile functionality appeared not so long ago, I have not use it yet, and have not upgraded the applications to actual versions. Should I do some actions to move to new version to keep all my data?

Thanks!

brocaar, 2017-11-20:
See https://forum.loraserver.io/t/release-lora-server-0-22-0-lora-app-server-0-14-x-read-carefully/350 :-) 

The reason to move this config option to LoRa Server is that network-configuration is in the network-server domain and should be configured / handled there. Also the network-server knows best about the available resources and limits like duty-cycles (eventually) an should be able to choose which receive window would be the most resource efficient choice for downlink.

alexanderplanar, 2017-11-20:
Thanks for clarifying, **brocaar**!
I will try to update application to new version. Let's see what will happen :)

alexanderplanar, 2017-11-20:
By the way, I'm from Russia, and I'm interesting the LoRa in our country.
I have read the post [()](https://github.com/brocaar/loraserver/issues/113)
I did not find the RU868 band at configuration page ([](https://docs.loraserver.io/install/configuration/)
What do you think about adding the band?

Thanks!

brocaar, 2017-11-20:
I will follow the LoRaWAN Regional Parameters specification by the LoRa Alliance. If it gets added there, I'll implement it :-)

alexanderplanar, 2017-11-21:
**Ok, I understand :-) Looking forward for the specification issuing.**

Sorry for offtopic.
Do you have any information about the date when the new revision of Regional Parameters will be issued? I found out that LoRaWAN Alliance going to issue new revision (B) of Regional Parameters til New Year. Do you know anything about this?

Thanks!

brocaar, 2017-11-21:
Sorry, I don't know.

closed 2017-11-21 by brocaar

alexanderplanar, 2017-11-21:
It seems to be work fine after upgrading.

sputnik-alarm, 2019-01-11:
In LoRa Server v2.4.1 we still have this issue, "rx2_dr=..." option not working, datarate is always DR_0.

brocaar, 2019-01-21:
Please note that changing the `rx2_dr` requires you to restart LoRa Server **and** 1) either the device to OTAA re-activate or 2) to acknowledge the mac-command that changes the RX2 parameters on the device. LoRa Server will keep using the "old" RX2 value until the device is made aware that this parameter changed.