brocaar/chirpstack-application-server #174: Add Bearer schema to HTTP Authorization header for REST API

fifthaxe, 2017-10-24:
Hi,

I am a bit confused by the way you are implementing JWT. AFAIK we are supposed to specify the Bearer schema while dealing with HTTP Authorization header.

In auth.getTokenFromContext function, you are fetching token directly from md["authorization"] whereas this should be a string matching "Bearer $token".

Is this a a REST proxy to gRPC limitation ?

Looking forward to getting your point of view.

pmiam999, 2017-10-25:
@fifthaxe : there is no such limitation : the HTTP **Authorization** and **Grpc-Metadata-Authorization** headers are mapped to **authorization** metadata and they can contain anything.

Adding the Bearer schema sounds a good improvement to me.

brocaar, 2017-10-25:
I'm afraid this is currently not possible, because of the gRPC gateway (exposing the gRPC methods as a RESTFul JSON api). 

https://github.com/grpc-ecosystem/grpc-gateway#mapping-grpc-to-http

> HTTP `Authorization` header is added as `authorization` gRPC request header

That is the reason why it is (currently) implemented that way.

brocaar, 2017-10-25:
Yes, you're right. Some more info: https://security.stackexchange.com/questions/108662/why-is-bearer-required-before-the-token-in-authorization-header-in-a-http-re.

Would one of you mind creating a pull-request for this? I think it is important to implement a fallback to the current implementation, to make sure the update does not break any existing implementations (maybe logging a "deprecation" warning)?

pmiam999, 2017-10-25:
I take the point and will raise a message => WARNING: Deprecated Authorization header : expected RFC1945 format : "Authorization: <type> <credentials>"

fifthaxe, 2017-10-25:
Wow so fast. Thanks guys !

brocaar, 2017-12-11:
Thanks all! This has been merged and released. I'm going to close this issue.

closed 2017-12-11 by brocaar

yinfangcai66, 2021-07-16:
11111