brocaar/chirpstack-application-server #177: Error rpc error: code = Unknown desc = sql: converting Exec argument #5's type: unsupported type backend.RatePolicy, a string (code: 2)

waytai, 2017-11-03:
### Is this a bug or a feature request?

### What did you expect?
config the service profile correct
### What happened?
can't config service profile
### What version are your using?
commit 2451e49b0e0764c4119ca0e9176a846e1c3909ac
### How can your issue be reproduced?

### Could you share your log output?

brocaar, 2017-11-03:
> How can your issue be reproduced?

Please share some information. I don't get this error and without any steps how it can be reproduced, it will be guessing for me. E.g.

* Did you compile the code yourself
* Which compiler version did you use
* Same for LoRa Server source?
* Did you upgrade of fresh installation
* ...

waytai, 2017-11-03:
lora-app-server_0.14.0_amd64.deb 
download from https://docs.loraserver.io/lora-app-server/overview/downloads/

loraserver_0.22.0_amd64.deb
download from https://docs.loraserver.io/loraserver/overview/downloads/
my go version
go version go1.7rc6 linux/amd64

and my postgres database is new

waytai, 2017-11-03:
It's ok now, thanks for you kindly reply

closed 2017-11-04 by waytai

waytai, 2017-11-06:
go version go1.9.2 linux/amd64
build is ok

mrostamii, 2020-04-08:
Check the pointer type of variable. I've had the same problem with my code and finally, the problem solved by the pointer type of a struct that called by another func.