brocaar/chirpstack-application-server #179: Organization admin user cannot get organization applications

muka, 2017-11-09:
From UI an user can be admin of an organization and create applications. Those are not listed then in the overview for the org admin, but show up for the global admin.

The relevant API call is `/api/applications?organizationID=5&limit=20&offset=0` which point respectively to
- https://github.com/brocaar/lora-app-server/blob/master/internal/api/application.go#L158
- https://github.com/brocaar/lora-app-server/blob/937afe84abe9c172ac0f684ed4f6748f3340e9ab/internal/storage/application.go#L160

brocaar, 2017-11-13:
Could you post some screenshots? I'm not sure if I fully understand the issue.

brocaar, 2017-11-13:
I just created an application as an "org admin". After creating this app, I can see it in the list of applications for this organization.

muka, 2017-11-13:
Hi @brocaar, you can reproduce easily:

1. login as global admin, create org, create app, create org admin
2. login as org admin, no apps available
3. as global admin add org admin to app users
4. org admin view the app in the list

Seems there is a kind of "ownership" enforcement on apps, may it be the case?

Thanks

brocaar, 2017-11-13:
Are you using the latest LoRa App Server / LoRa Server version? https://forum.loraserver.io/t/release-lora-server-0-22-0-lora-app-server-0-14-x-read-carefully/350

muka, 2017-11-13:
No you are right, I miss the release notes.. I am still somewhere back in time (`0.13.1-1-g469930b`)

I will try again once up to 0.14, closing the issue I will reopen if still apply.

Thank you

closed 2017-11-13 by muka

muka, 2017-11-14:
Tested with last tag (~0.14) and all works fine. Great work!