brocaar/chirpstack-application-server #180: Decrypt frames in frame log

x0y1z2, 2017-11-09:
In some cases it is useful to look at the decrypted contents of the packages.
This feature need for debugging.

brocaar, 2017-11-09:
Please see https://docs.loraserver.io/lora-app-server/integrate/data/ for subscribing to the decrypted data.

x0y1z2, 2017-11-10:
I know about this feature, and use it for my application. But sometimes I need to have full log of data exchange for debugging my application and devices (with ADR and OTAA packages). In web interface I can view log, but I want to see payload of packets, but web interface show only encrypted packets, and don't show content of JoinAccept packets (CFList and other fields).
I don't know go programming language, but I think, that this is easy to add this feature, because your code contains algorithms for decrypting data.

brocaar, 2017-11-13:
This is a bit tricky, as the logging is performed by the LoRa Server component, which does not have all the keys to decrypt. Also please note that a join-accept is encrypted using the AppKey. When using a third-party join-server, non of the LoRa Server components have the key to decrypt the join-accept message.

rick-pit, 2017-12-11:
Im confused by this too. i like to access to this log in any kind of emergency where i have to rebuilt data, or in case of comunication failure between aplication server and lora server where i have to retrieve manually frames.

brocaar, 2017-12-12:
I don't this feature will be possible. As I mentioned above, the keys are distributed across different components. E.g. when using a join-server, there is no way to decrypt a join-accept message. Not for LoRa Server, not for LoRa App Server.

You really should not use the raw frames for application integration!

brocaar, 2018-04-09:
This has been implemented since LoRa App Server 0.19.0.

closed 2018-04-09 by brocaar