brocaar/chirpstack-application-server #181: [Feature] Ability to move nodes/applications.

mihaispataru, 2017-11-13:
**Is this a bug or a feature request?**

Feature

**What did you expect?**

1. Possibility to move nodes from an application to another as admin or as user from an organization
2. Possibility to move applications with nodes from an organization to another as admin.

brocaar, 2017-11-13:
I'm not sure if this is worth the complexity. There is more complexity than just changing the foreign-keys. Devices have a dependency on a device-profile and applications on a service-profile. The service-profile, device-profile and device **must** be provisioned on the same network-server.

brocaar, 2019-07-02:
This is possible using the API, see: https://github.com/brocaar/lora-app-server/blob/master/api/device.proto#L139.

Given the complexity of validation, I do not have plans to support this in the web-interface.

closed 2019-07-02 by brocaar