brocaar/chirpstack-application-server #188: compiling error v 0.15.0

lsdev75, 2017-12-13:
### Is this a bug or a feature request?
bug
### What did you expect?
Compilation of lora-server-app
### What happened?
Error during compile: it doesn't find the google.golang.org/genproto/googleapis/google/api/annotations/  neither in vendor nor in $GOPATH
In vendor there is the following directory: 
google.golang.org/genproto/googleapis/api/annotations/
### What version are your using?
the last one 0.15.0
### How can your issue be reproduced?

### Could you share your log output?
package google.golang.org/genproto/googleapis/google/api/annotations/not found neither in vendor nor in $GOPATH

Creating the google as parent of api directry it compiles

brocaar, 2017-12-13:
Please try to compile using the provided `Dockerfile-devel` Docker environment. That way we're both sure you have all the requirements installed & at the right versions.

If there really would be dependencies missing, I would be unable to release as the build server would raise the same error.

lsdev75, 2017-12-13:
I'm not so familiar with docker, but I'll try later.
Now I'm looking on my environment variables and paths.

closed 2018-01-15 by brocaar