brocaar/chirpstack-application-server #189: rxWindow choice seems removed in latest version

borisbstyle, 2017-12-15:
### Is this a bug or a feature request?
bug
### What did you expect?
feature to work as in earlier versions
### What happened?
configuration for rxWindow dissapeared. By default only RX1 active
### What version are your using?
latestexit
### How can your issue be reproduced?
0.15.0
### Could you share your log output?
Option for choosing RX1 or RX2 is simply missing in device profile configuration

wero1414, 2017-12-15:
Same problem here we want to use the RX1 instead of RX2 for class C but now we cant, hope someone know how to configure the Rx windows

brocaar, 2017-12-16:
For Class-A it is up to the network-server to decide to use RX1 or RX2 (based on timing, gateway availability etc...). This used to be a config up which could be set by the user, but this has been dropped to stay consistent with the device- and service- profiles as defined by the LoRaWAN Backend Interfaces specification.

@wero1414 please note that when you check the Class-C option in the device-profile, the network-server will automatically use RX2 parameters.

borisbstyle, 2017-12-18:
@brocaar I can't remember finding this in release notes?

How can I verify that RX2 is really working?

borisbstyle, 2017-12-20:
Is there a way to force RX2 being used first in the current version? I am seeing significant performance degradation in downlink since the last version, just because RX1 being used. I would love to use RX2 if somehow possible?

borisbstyle, 2017-12-20:
Also what is the best downgrade procedure for lora packages? Are there any manuals for that?

brocaar, 2017-12-22:
Currently this is not possible anymore. Would it help when this becomes a global config option in LoRa Server (default: auto, options: auto, rx1, rx2)? My worry with making this available to the end-user is that it gives the end-user the power to always use RX2, which will make your network less efficient (usually a lower data-rate + fixed frequency).

Letting the network-server decide which RX slot to use means the network can be used more efficiently.

brocaar, 2018-08-01:
Closing this issue. The RX1 / RX2 option won't be added to LoRa App Server as this should be handled in the domain of the network-provider, not the end-user (as giving the end-user the power to always use RX2 could impact the network performance).

I am still planning to add support for this in LoRa Server. Either using a fall-back on RX2 when RX1 scheduling fails. And maybe also with an option to hardcode RX1 / RX2 scheduling in the configuration file.

closed 2018-08-01 by brocaar