brocaar/chirpstack-application-server #190: Can't create/edit/retrieve Device Profiles unless attached to NetworkServer

jcampanell-cablelabs, 2017-12-20:
### Is this a bug or a feature request?
I will call it a bug, as no error is reported if the NetworkServer is not present.

### What did you expect?
I expected that in creating a Device Profile, that Device Profile could then be loaded to the UI screen and edited.

### What happened?
Device Profile UI is basically blank.

### What version are your using?
Latest from cloned repo, 12/19, built myself.

### How can your issue be reproduced?
With a bogus NetworkServer, create a DeviceProfile.  Open via UI once created.

### Could you share your log output?
No log output of relevance that I noticed.

I understand that the DeviceProfile and ServiceProfile data is stored on the LoraServer.  However, if the LoraServer is down or non-existent, the LoraAppServer should probably keep a local copy of the data so it can be retrieved and updated, and perhaps scheduled for a push at a later time.  Where LoraAppServer and LoraServer used to be more dependent on each other in a 1-to-1 relationship, the new model of having any number of LoraServers should probably mean that LoraAppServer steps up to be a "database of record", and pushes the data to the LoraServers as they become available if they are not available at data entry time.  Also note that if a LoraServer is decommissioned, the DeviceProfile and ServiceProfile data would be lost.

brocaar, 2017-12-22:
Would it help when I show a warning in this case. E.g. the user wants to create a device-profile in the UI, and when no network-server is available a warning is shown that first a service-profile must be created (through which an organization is linked to a network-server).

The same could be valid that when an user wants to create a service-profile and no network-server exists, it warns the user that it must first create a network-server.

jcampanell-cablelabs, 2017-12-22:
The error/warning messages would technically work, but the net result is that the system appears to be pretty fragile if something goes down (loraserver crash, loraserver server restart, network down, etc.).  I guess if the systems are all up and running, which should be the vast majority of the time, it basically works.  But if the loraserver(s) isn't up, or one goes down permanently, things get into a pretty ugly state.  I mean, if I go through a lot of effort to get my DeviceProfile *just right*, and then the loraserver gets decommissioned (hacked/shut down by the company, etc.), then my DeviceProfile settings are lost since there is no local copy on lora-app-server.  As a user, that would be frustrating.

closed 2019-07-02 by brocaar