brocaar/chirpstack-application-server #191: Add logging of all queries and responses and all errors.

jcampanell-cablelabs, 2017-12-20:
### Is this a bug or a feature request?
Feature Request

### What did you expect?
Queries made to the server to show up in the logs, as well as responses and more logging on errors.

### What happened?
Minimal logging occurs on success cases.

### What version are your using?
Latest build from source, retrieved 12/19.

### How can your issue be reproduced?
Use the system.  Especially when using the REST API.  In particular, I was working on creating new users via REST, and the newer code was rejecting the request.  The log showed no indication that the query was seen, let alone that there was any kind of error.

### Could you share your log output?
Nothing to share.

As a developer using the REST API, checking the server logs is usually the easiest way to see how I am screwing up my calls.  I don't even get an indication that the request is seen by the server, let alone that an error occurred.

brocaar, 2017-12-22:
Unfortunately this is not something that is supported by the Go database package or PostgreSQL driver itself. I will look into how we can do this in a generic way, without the need to add a manual log around every query in the code.

brocaar, 2018-01-15:
In my previous comment I misinterpreted "query" as database queries (which I think would be useful too to log in case of debug logging). I found a fairly easy way to do this and implemented this already in LoRa Server: https://github.com/brocaar/loraserver/commit/a4c5652f84f492cac4d11f4220a6307da5aa4035

I will add this too to LoRa App Server and will go through the the code to see where extra (debug) logging would be needed. I agree that more logging is beneficial :-)

brocaar, 2018-01-16:
All request are now logged as info / warning / error level (depending the gRPC return code).

commit a3ca857b (2018-01-16): Implement gRPC server logging.

Closes #191.
closed 2018-01-16 by brocaar