brocaar/chirpstack-application-server #192: Service Profile cannot be deleted from UI

jcampanell-cablelabs, 2017-12-21:
### Is this a bug or a feature request?
Bug

### What did you expect?
To have a button that deletes the Service Profile

### What happened?
There is instead another button to delete the Organization.  And it deletes the Organization.

### What version are your using?
Latest pull was on 12/19 - built myself

### How can your issue be reproduced?
Go to organization page, select Service Profiles and pull on up.  Normal placement for delete button is top-right of list.  DELETE ORGANIZATION is there instead (see device profiles for expected layout). 

### Could you share your log output?
None of relevance.

brocaar, 2017-12-22:
It should display two buttons, one to remove the organization (as the service-profile view is rendered within the [OrganizationLayout](https://github.com/brocaar/lora-app-server/blob/master/ui/src/views/organizations/OrganizationLayout.js), and one delete service-profile button within the service-profile form, see below:

[image]

jcampanell-cablelabs, 2017-12-22:
My mistake then.  Issue is that the Service profile is not being loaded, and so the REMOVE SERVICE PROFILE button doesn't display.  Working for you, so perhaps my build went awry.  Will close.

closed 2017-12-22 by jcampanell-cablelabs