brocaar/chirpstack-application-server #196: Possibility to add organization-name in front of MQTT topic.

bramkennis, 2018-01-09:
It would be a nice feature, to have the organization-name in front of the MQTT topic of the published data-up packets. 
Current syntax:
**application/2/node/0004a30b001b1d62/rx**
New syntax:
**organization-1/application/2/node/0004a30b001b1d62/rx**

It makes it easier to let customers have access to only their root topics.

halukmy, 2018-01-09:
yes you can

bramkennis, 2018-01-11:
Great, any idea how? Or do i need to change the mqtthandler file and rebuild the application:

```
func (h *MQTTHandler) SendDataUp(payload handler.DataUpPayload) error {
	b, err := json.Marshal(payload)
	if err != nil {
		return fmt.Errorf("handler/mqtt: data-up payload marshal error: %s", err)
	}

	topic := fmt.Sprintf("**organization**/application/%d/node/%s/rx", payload.ApplicationID, payload.DevEUI)
	log.WithField("topic", topic).Info("handler/mqtt: publishing data-up payload")
	if token := h.conn.Publish(topic, 0, false, b); token.Wait() && token.Error() != nil {
		return fmt.Errorf("handler/mqtt: publish data-up payload error: %s", err)
	}
	return nil
}
```

An option to make this dynamically would be great.

brocaar, 2018-01-11:
Probably you need to add and populate an `OrganizationID` to the `DataUpPayload` struct and then use it like the application id. Not that this makes it easier, but this is currently already possible, see: https://docs.loraserver.io/install/mqtt-auth/

bramkennis, 2018-01-11:
Thanks i will look in to that and give feedback.

brocaar, 2018-01-15:
Actually I might have an usecase myself where it would be helpful to make this configurable. E.g. maybe this could be config value representing a text template (https://golang.org/pkg/text/template/). 

Then in your case the value would be:

`organization/{{.OrganizationID}}/application/{{.ApplicationID}}/device/{{.DevEUI}}`

The rendered value would then act as the prefix used for the `.../rx`, `.../tx`, `.../ack` topics..

brocaar, 2018-08-01:
Is this still a feature request? Currently some form of topic template is already supported. Please note that adding additional details to the topic template has some performance impact as more DB queries are required to fetch the application and organization details. Is with worth it?

Note that MQTT authentication / authorization can already be setup to perform authentication on application id level (see also: https://www.loraserver.io/install/mqtt-auth/).

bramkennis, 2018-08-02:
> Beantwoorden - Aan: brocaar/lora-app-server <reply@reply.github.com>

Hi Orne,

This is no feature request anymore. We will use the MQTT authentication to get access to the right application and topic.

Met vriendelijke groet
Bram Kennis

CTO
[ttp://mailimages.oac.be/sig/logo.png]<http://www.oac.be/>     [ttp://mailimages.oac.be/sig/lora-rgb.png]
[ttp://mailimages.oac.be/sig/tel.png]+32 14 / 55 83 83
[ttp://mailimages.oac.be/sig/fax.png]+32 14 / 55 24 58
[ttp://mailimages.oac.be/sig/eml.png]bram@oac.be<mailto:bram@oac.be>
[ttp://mailimages.oac.be/sig/www.png]www.oac.be<http://www.oac.be>
[ttp://mailimages.oac.be/sig/asn.png]AS56675

DISCLAIMER
This e-mail and any attached files may contain confidential information and may also be privileged. It is intended only for the use of the recipient named above. If you are not the intended recipient, any disclosure, reproduction, copying, distribution or other use of this communication is strictly prohibited. If you have received this transmission in error please notify the sender immediately and then delete this e-mail.
The integrity and security of e-mail transmission cannot be guaranteed. The sender is therefore in no way liable for any errors or omissions in the content of this message, which may arise as a result of e-mail transmission. If verification is required, please request a hard copy.

[ttp://mailimages.oac.be/sig/env.png]Please consider the environment before printing this e-mail.

closed 2018-08-06 by brocaar