brocaar/chirpstack-application-server #197: Error invalid channel (code:3)

bramkennis, 2018-01-11:
### Is this a bug or a feature request?
Bug?

### What did you expect?
Correct way of adding extra channels.

### What happened?
configuring extra channels

### What version are your using?
Last versions

### How can your issue be reproduced?
adding channel 3,4,5,6,7 and hit submit.

### Could you share your log output?
[image]
[image]

brocaar, 2018-01-11:
For EU the only channels that are defined by the LoRaWAN Regional Parameters are 0, 1, 2. The rest are the extra channels which you already entered below.

bramkennis, 2018-01-11:
Ok great, so now i just need to install the lora-channel-manager on the multitech gateways to read and update the config file?

brocaar, 2018-01-11:
correct

closed 2018-01-15 by brocaar