brocaar/chirpstack-application-server #205: Database is slow for packet/frame data?

PhatHub, 2018-01-23:
### Is this a bug or a feature request?
Bug

### What did you expect?
I was expecting the database to load a little bit faster...

### What happened?
The database is taking several minutes to load data. The client is having issues with RestfulAPI responses, and the admin interface takes about 2 minutes to reveal Frame Logs.

### What version are your using?
0.12.0

### How can your issue be reproduced?
I make Restful API requests for Frame Data.
I use the Admin page for Frame Data.

### Could you share your log output?
I will if requested and commands are specified! :)

Also I guess a feature request is on my mind: 
If it's possibly due to too many packets in the database, is there a way to purge older data?

brocaar, 2018-01-24:
I'm not sure if this is a bug (missing index or inefficient query which is not using the indices properly), or if this is hardware related (e.g. slow storage, not enough RAM, ....). Would you be able to debug this? I'll also look into this.

Are you sure you're running 0.12.0? In that feature the frame logging wasn't implemented yet ;-)

You can purge old `frame_log` records the same way as you can purge "old" gateway statistics data. This won't break any functionality. See for an example: https://docs.loraserver.io/loraserver/use/gateway-management/.

E.g.

```sql
delete from frame_log where created_at < now() - interval '1 week';
```

brocaar, 2018-01-24:
You can analyze how PostgreSQL execute the queries by using `explain`. Example:

```sql
explain select * from frame_log where dev_eui = decode('01010101010101', 'hex') order by created_at desc limit 20 offset 0;
```

(this is the same query as used to fetch the frame-logs: https://github.com/brocaar/loraserver/blob/master/internal/node/frame_log.go#L49).

But it could be that the size of the table (and indices on `dev_eui` and `created_at`) are pushing the limits of the machine on which PostgreSQL is running. In that case you should run the above query to truncate the table so that it only stores the data of last week.

Note: the mentioned tables are in the LoRa Server database, not LoRa App Server!

PhatHub, 2018-01-26:
@brocaar Thank you! 
* Version: 
    * Well, the lora-app-server's `config.toml` file states that the version parameter is "0.12.0"
    * Loraserver's `config.toml` is version "0.18.0"
* I will definitely look into this. ;) The lag in the frame info was around the time two things happened, although I didn't think either actions would affect performance like this:
    * I did a pull from the repos to update my code
    * A friend migrated the server from a low-capacity flash drive to a higher-capacity spinning drive. I don't think the bare metal had changed.
* So this is why I also wanted to know how to purge old frame data.

I've also noticed that it's taking a bit of time in the web interface to populate the "Node Activation" fields... I'll have to look more into the system and see what's happening.

PhatHub, 2018-01-30:
Okay, Postgres is definitely choking a bit. 
"top" is reporting postmaster as grabbing 4.7% of the CPU and 7.8% of memory.
...Then top crashed and I lost my SSH connection.  :cold_sweat:  :stuck_out_tongue_closed_eyes:
I'll have to reboot the VM and maybe configure for more CPU cores.

iegomez, 2018-01-30:
You mentioned that a friend migrated your server, so it's possible that he dumped and restored your DB, but missed the indexes. You can check specific indexes for the frame_log table like this (being connected to the loraserver DB, of course):

```sql
\d+ frame_log
```

If everything is right, you'd have something like this (it could differ a bit between versions), the important part being the indexes on created_at and dev_eui:

```
                                                           Table "public.frame_log"
   Column    |           Type           | Collation | Nullable |                Default                | Storage  | Stats target | Description 
-------------+--------------------------+-----------+----------+---------------------------------------+----------+--------------+-------------
 id          | bigint                   |           | not null | nextval('frame_log_id_seq'::regclass) | plain    |              | 
 created_at  | timestamp with time zone |           | not null |                                       | plain    |              | 
 dev_eui     | bytea                    |           | not null |                                       | extended |              | 
 rx_info_set | jsonb                    |           |          |                                       | extended |              | 
 tx_info     | jsonb                    |           |          |                                       | extended |              | 
 phy_payload | bytea                    |           | not null |                                       | extended |              | 
Indexes:
    "frame_log_pkey" PRIMARY KEY, btree (id)
    "idx_frame_log_created_at" btree (created_at)
    "idx_frame_log_dev_eui" btree (dev_eui)
```

 Also, could you run a simple count to check how many frame logs you have stored? I.e.:

```sql
select count(id) from frame_log;
```

brocaar, 2018-01-30:
>  4.7% of the CPU and 7.8% of memory.

That doesn't sound like a reason to me to add more CPU cores?

FYI, I've received some other concerns from other companies with regards to the frame-logging feature. On large networks, this would fill up the database rapidly.

I'm considering to re-implement this as a streaming feature. E.g. when you're interested in receiving the frame-logs for a device (and possibly gateway), you can subscribe to it and you'll see it, but it will not be stored in the database anymore.

From the web-interface you would then start seeing frames when opening the frame-log tab. Ideally live :-)

PhatHub, 2018-01-30:
@iegomez : Thanks for the hint, but actually he just cloned the entire VM, so it's the same OS image but just moved from flash to spinning media (for the extra storage space).

@brocaar : Actually, maybe a user-settable history cap? Setting a max number of packets to keep in storage and keep accessible for the RESTful API? Due to my "playing" with Lorawan, I have several months of data with incredibly high rates of transmissions, and I probably only need about two to four week's worth of data to query against.  
To your point: Considering that the payload's readily decrypted at the MQTT subscription but not at the web interface, the frame log display is probably of limited use aside from seeing that packets had been received while the app developer/admin was out to lunch.

I'm recently noticing that my nodes' authentication data/keys are taking a little while to display on the admin interface too... at first I thought that my AppSKey and NwkSKey being zeroes were due to my laziness (of not wanting to input a huge hex string into my Arduinos, which I did at first) but then the actual hex values popped up after a while. 
So this makes me think: If my keys aren't popping up in time in the admin service, could the server be dropping packets if it can't verify the MIC? Or is that a different process?

Anyway, I'll work with my friend on diagnosing it further. Maybe I'll ask for another core while I'm at it, and play with shutting down unrelated (to the core loraserver/lora app server/lora bridge) services.

brocaar, 2018-02-06:
@PhatHub the next version will not keep a time limited cap. Instead nothing will be stored to the database anymore, but you will be able to live-stream the frames sent / received by your device or gateway from the UI.

brocaar, 2018-02-08:
@PhatHub I'm currently releasing a new version of LoRa Server and LoRa App Server with the new streaming frame-logging. You should keep the `frame_log` table in your LoRa Server database, but you can truncate it. Functionally it won't be used anymore, but please don't remove it as I'll remove this table programmatically in the next release :-)

Potentially this could free up a lot of resources on your server.

brocaar, 2018-02-28:
The new frame-logging feature has been released and data is not stored in the DB anymore.

closed 2018-02-28 by brocaar

PhatHub, 2018-03-05:
Okay, sounds good. I'll check it out.