brocaar/chirpstack-application-server #213: Reset Frame counter for ABP devices

espuser, 2018-02-04:
### Is this a bug or a feature request?
I know that the frame counter is part of a security feature. For development it is heavy to wait until the frame counter is above the last count before Device reset.

### What did you expect?
a button to reset the frame counter
### What happened?

### What version are your using?
latest version downloaded 3 days ago for Debian jessie
### How can your issue be reproduced?

### Could you share your log output?

brocaar, 2018-02-04:
You can either use the relax frame-counter mode (which is not secure) or re-activate your device (ABP) with the same keys and frame-counters to `0` (which isn't secure either).

closed 2018-02-04 by brocaar

ayodhyadhameesha, 2022-08-16:
Hello brocaar,
Can you help me to reactivate the device automatically, like from a code or something.