brocaar/chirpstack-application-server #214: Unable to create a network server

dpmendes, 2018-02-08:
I believe this bug is probably a misconfiguration I made somewhere, but I can't find a way to fix it.
On the network servers screen, I add a network-server name and a network-server server (localhost:8000). All the other fields are blank. When I click "submit", I receive the error "Error context deadline exceeded (code: 2)", as shown next:

[image]

I'm running lora-app-server version 0.17.1.
Everything else is working fine since I can connect an end device to my gateway and see its data on a TTN app server.

Regards.

brocaar, 2018-02-08:
The forum is a better place for support questions: https://forum.loraserver.io/t/can-not-register-network-server/407.

closed 2018-02-08 by brocaar

dpmendes, 2018-02-08:
Thanks.

Sorry for posting on the wrong place.