brocaar/chirpstack-application-server #216: Application EUI field is missing?

JairoRotava, 2018-02-16:
### Is this a bug or a feature request?
bug
### What did you expect?
set or get the appeui from somewhere in the lora-app-serve
### What happened?
after upgrade to the version 0.18.0 I don't find appeui anymore
### What version are your using?
lora-app-server: 0.18.0
loraserver: 0.24.0
### How can your issue be reproduced?
Add a new node, use OTAA. 
### Could you share your log output?

brocaar, 2018-02-18:
It is not needed. From the device the AppEUI / JoinEUI (it got renamed in LoRaWAN 1.1) has the purpose to signal the network-server which join-server to use to authenticate the join-request.

closed 2018-02-18 by brocaar

JairoRotava, 2018-02-19:
I missed that. Now I can use a random AppEUI/JoinEUI and the node joins and communicate just fine. 
Very good.